LinuxCNC/linuxcnc #443: Lathe configuration error using the wizard

Masznotwor, 2018-06-17:
I installed LinuxCNC 2.7.13
I made the lathe configurations in the Stepcnof wizard (I know how well because I already configure 4 lathe)
You will pass the test drive to the Z axis and X axis work
After the wizard completes and attempts to run, it gets an error
ebug file information:
.
./test.hal:61: parameter or pin 'stepgen.2.position-scale' not found
11941
   PID TTY STAT TIME COMMAND
Stopping realtime threads
Unloading hal components

Old configurations made on previous versions of linuxcnc from 2015 work well so the new configuration wizard is guilty

andypugh, 2018-06-18:
I believe that this is already fixed in  6a1265c3445489e38079d93ff145a6fdec78d7bd which will be in the next release. 

Thanks for the report. 

As a workaround you can edit the HAL file manually (and you probably already have)

andypugh, 2018-06-18:
Looking again. This is tagged as already in 2.7.13 and the changes do not extend to stepgen numbering.

SebKuzminsky, 2018-06-18:
I think this was fixed in 552b4304, which came after 2.7.13 and hasn't been released yet.

SebKuzminsky, 2018-06-18:
Yeah, that commit fixes it.  Looks like it's time for 2.7.14!

SebKuzminsky, 2018-06-18:
This has been fixed in the just-released 2.7.14.  Thanks for reporting the problem!

closed 2018-06-18 by SebKuzminsky