LinuxCNC/linuxcnc #444: PWM is not the same for M4 and M3

Masznotwor, 2018-06-19:
hi (thank you for repairing the previous problem, however, a new one appeared)

**## Here are the steps I follow to reproduce the issue:**

I made configurations in the wizard - Lathe
I start and base the machine
in MDI it specifies M3 S100 or M4 S100 commands also similar

**## This is what I expected to happen:**

I should get the same PWM for right and left turns.

**## This is what happened instead:**
For M3 PWM seems OK but for M4 it is very slow
of course, if the speed increases, PWM for M4 and M3 increases, but M4 remains much slower

[VIDEO TEST](https://youtu.be/Sf0e-dehbwE)

**## It worked properly before this:**
I made my previous configurations a few years ago and always operated from previous years

**## Information about my hardware and software:**

Problem I detected LinuxCNC 2.7.13 released then I updated to LinuxCNC 2.7.14 released I released configurations but the problem remained

SebKuzminsky, 2018-06-19:
Yep, this looks like another bug.

StepConf configures the spindle pwmgen with a small positive offset, and the pwmgen component (incorrectly, i think) always *adds* this offset to the duty cycle.  So negative values on the `.value` pin are closer to zero than you'd think, just like your video so clearly shows.

The [pwmgen manpage](http://linuxcnc.org/docs/2.7/html/man/man9/pwmgen.9.html) claims to "add/subtract" to offset from the computed duty cycle, presumably following the sign of the `.value` pin.  (That manpage also incorrectly lists many pwmgen pins as params, but we converted them to pins back in 2008...)

`git log` shows that pwmgen has always behaved in this buggy way, and it took a careful observer like @Masznotwor to spot the problem.  This may mean that the pwmgen component isn't widely used.  Maybe most folks use Mesa hardware if they need pwm outputs?

SebKuzminsky, 2018-06-19:
Nope, I misread "underlined +" as the plus/minus symbol "±".  The pwmgen manpage *does* say that the offset is always *added* to the duty-cycle, whether the `.value` pin is positive or negative.

I think the pwmgen offset is intended to avoid the dead zone around 0% duty cycle, and for that reason i think the offset should be *subtracted* from the duty cycle when the input value is negative.

Am I missing some other use of the offset?

For now I think @Masznotwor (and anyone with a pwmgen-controlled spindle) would be best served by setting the offset to 0 and commanding a slightly higher S-word to spin the spindle up.

andypugh, 2018-06-19:
Offset could be used to balance an unbalanced device. I think that "offset" probably should be added regardless of the sign of value. The problem may be that stepgen is attempting to use it as a "deadband" instead. 

The offset in PID is useful for balancing the weight of the head / table on a mill. I don't know if there are applications where it would, instead, be appropriate to apply it to the PWMgen.

Masznotwor, 2018-06-19:
Do not be afraid that I know English poorly, I also understand what is going on in HAL and INI files, so I can be wrong:
I think that it should be done simply:
the component generates PWM and that's it
the direction of rotation is not to influence PWM
And if it is something you need, it should rather be run as an option if it causes problems

I think that many people use one-way spindles (water spindle) and only use PWM CW

I had not noticed the error but the previous lathes had an encoder and it may have affected. I do not know, I just did not go down that low speed

here the problem seems to be solved, unfortunately I do not know how to apply it and whether it is a good solution
https://forum.linuxcnc.org/49-basic-configuration/28700-lpt-spindel-pwm-different-speed-with-m3-and-m4

I have such ingredients

loadrt pwmgen output_type=1

addf parport.0.read base-thread
addf stepgen.make-pulses base-thread
addf pwmgen.make-pulses base-thread
addf parport.0.write base-thread
addf parport.0.reset base-thread

addf stepgen.capture-position servo-thread
addf motion-command-handler servo-thread
addf motion-controller servo-thread
addf stepgen.update-freq servo-thread
addf pwmgen.update servo-thread

net spindle-cmd-rpm => pwmgen.0.value
net spindle-on <= motion.spindle-on => pwmgen.0.enable
net spindle-pwm <= pwmgen.0.pwm
setp pwmgen.0.pwm-freq 100.0
setp pwmgen.0.scale 2984.21052632
setp pwmgen.0.offset 0.05
setp pwmgen.0.dither-pwm true
net spindle-cmd-rpm     <= motion.spindle-speed-out
net spindle-cmd-rpm-abs <= motion.spindle-speed-out-abs
net spindle-cmd-rps     <= motion.spindle-speed-out-rps
net spindle-cmd-rps-abs <= motion.spindle-speed-out-rps-abs
net spindle-at-speed    => motion.spindle-at-speed
net spindle-ccw <= motion.spindle-reverse

I actually checked, you can not notice at higher revolutions because then the m3 / m4 differential is about 1V on the transducer