LinuxCNC/linuxcnc #445: Glade editor on Stetch iso not newest and is buggy

c-morley, 2018-06-19:
The version of GLADE on the debian ISO is 3.8.0
it has been reported (and confirmed by me ) as buggy/crashy.
https://forum.linuxcnc.org/41-guis/34759-custom-gui-with-glade#112582
(Not the first complaint)

IIRC the newest version of GLADE for gtk2 is 3.8.6
It is not so buggy.
We should try to use the newest version on the ISO.

SebKuzminsky, 2018-06-19:
The version of glade in Stretch is 3.20.0 (though I don't know enough about glade to :
https://packages.debian.org/stretch/glade

We generally don't update our install media very often, instead it is our recommendation that the user should update their installed machines using the system's package manager, first immediately after installing, then periodically after that.

c-morley, 2018-06-19:
it's unfortunate GLADE choose to use the numbering it did...
Any GLADE newer then  3.8.x is for GTK3 and is unusable with GTK2 widgets.
GLADE 3.20 can be added beside 3.8.x so it's no problem to have it on the system.

But gladevcp is pretty useless without an editor included for it.
I believe we have backported an Ubuntu deb package for it - we just aren't using the latest.
I think the bar is too high for the average user to add the deb themselves.
It's an ongoing/constant support question in the forums etc.

SebKuzminsky, 2018-06-19:
Thanks for educating me on that.

I'm not able to find a debian or ubuntu source package for glade 3.8 anywhere.

Ubuntu has packages starting with 3.12: http://archive.ubuntu.com/ubuntu/pool/main/g/glade/

Debian has packages for 3.6.7 and 3.9.1 (and earlier and later), but no 3.8: https://snapshot.debian.org/binary/glade/

So I don't know of a good easy solution here.  I'm open to suggestions.

c-morley, 2018-06-19:
I just remembered, did some investigating about this in this forum thread:
https://forum.linuxcnc.org/41-guis/34646-ui-development-environment-setup-help#111371

basically add a deb meant for wheezy and removing libgladeui-2-6
I ran out of time to get stretch version to work (if it's possible)
Is there a way to see where the version included with our ISO came from?
Well I'm assuming it came with it - the user indicated it did.
He was using 32bit and I use 64bit so there could be some diffeences

gmoccapy, 2018-08-04:
I posted a HowTo install glade 3.8.6 with 64 bit system. There is no package, so glade must be build from source. 

See: [https://forum.linuxcnc.org/48-gladevcp/33309-glade-3-20-will-not-work-with-linuxcnc-please-follow-this-for-3-8-6?start=10#99607](url)

KimK, 2018-08-04:
(above link adjusted for convenience)
@gmoccapy posted: [HowTo install glade 3.8.6 with 64 bit system.](https://forum.linuxcnc.org/48-gladevcp/33309-glade-3-20-will-not-work-with-linuxcnc-please-follow-this-for-3-8-6?start=10#99607) (links to LinuxCNC forum)

simaoamorim, 2019-11-29:
I know this is an old issue, but I guess it could be relevant for the porting to Stretch, why the big attachment to GTK2? It's 2019, why not start thinking about porting the widgets to GTK3? I guess the Glade problem would be automatically fixed with that...

c-morley, 2019-11-29:
linuxcnc currently can only use python 2. It's a huge technical problem, with few developers.
There is a gtk3 branch started - got stuck with the gcode graphics widget. And finally at the time GTK3 was not stable - they broke things every new release.
But yes Glade wold be fixed for new adopters.

simaoamorim, 2019-11-30:
Ok, so this is related to #403, right?
About porting the python modules to v3, I would like to help, I have a minimal
experience in python but I don't really know where to start or what exactly to do...