LinuxCNC/linuxcnc #447: unnecessary TP slowdown on some arc-to-line transitions

SebKuzminsky, 2018-06-21:
Originally reported here: https://www.cnczone.com/forums/tormach-pathpilot-/354884-tormach.html

The g-code in comment #6, when run on a 2.7.13 sim/axis machine, produces this result:
[image]
Note how machine velocity drops to 0 on exit from the first arc.

The partial slowdowns on entry to the first arc and exit from the second arc are expected, since the tool path makes sharp corners there, but the transition from the first arc to the following straight-feed looks tangent (or at least nearly so), and with G64 in effect the speed should not drop to 0.

This problem is reproducible on 2.7 and master.

This g-code demonstrates the problem:
~~~~~
; from https://www.cnczone.com/forums/tormach-pathpilot-/354884-tormach.html
; comment #6

g64   ; best speed, tolerances be damned
g91.1 ; incremental arc centers

;N1710 T9 G43 H9 M6
(0.500 End Mill HSS)
N1720 S3450 M3 M8
N1730 G0 X1.6009 Y0.7267
N1740 Z0.15
N1750 Z0.1075
N1760 G1 Z-0.215 F20.
N1770 G3 X1.5 Y0.35 Z-0.215 I0.6525 J-0.3767
N1780 G1 Y-0.35
N1790 G3 X1.5647 Y-0.5915 Z-0.215 I0.483 J0.
N1800 G0 Z0.15

m2
~~~~~

SebKuzminsky, 2018-06-22:
Here's a halscope trace that shows the same slowdown on an arc-arc transition (from the arcspiral.ngc example included in nc_files).
[image]
I triggered the scope on the first arc in the program.  The `motion.program-line` trace shows where the arcs start and stop.  The slowdown happens at the end of the first arc, just like in the example above.
Again, the blip at the start of the first arc is not a problem, it's due to a sharp corner in the tool path.

rodw-au, 2018-07-09:
Is this related to Issue #426 ?

We think this is causing problems with plasma machines as the slowdown/stop causes the plasma arc to burn a divot at that point because kerf width is a function of travel speed. Its hard to isolate as users noticing it have large, complex nests covering a full sheet of steel but somebody is working on it and will report an example when he gets a chance.

rene-dev, 2018-10-17:
https://github.com/LinuxCNC/linuxcnc/pull/509

closed 2018-11-06 by rene-dev

rodw-au, 2018-11-06:
On this forum thread 
https://forum.linuxcnc.org/plasma-laser/35449-another-plasma-component?start=30#120090
a user reports that he can no longer merge the reverse run branch into master following deployment of this fix. It would be great if the incompatibility could be resolved for the plasma nerds out here....

PR #512 (2018-11-03): Rellenberg/bug447 fix blend tangent fallback
should fix #447 and #426
commit 1fcb2610 (2018-11-03): tp: Fix fallback to parablic blending when tangent blend fails.

Fixes #447, which was caused by the TP not correctly falling back to
parabolic blending in certain cases. There is now logic to choose either
arc blending, treating segments as tangent, or parabolic blending,
depending on which is estimated to be fastest. This way, we should avoid
corner cases, since the performance check is always done when a new
motion is queued.