LinuxCNC/linuxcnc #448: Slave Axis motor not running in stepconf>test page

wbspathum, 2018-06-22:
i have two motors with two separate drivers for the Y axis. in stepconf > parallel port page i set 

Pin 4: Y Step
Pin 5: Y Direction
AND
Pin 8: Y Step
Pin 9:  Y Direction

but when i test it from the Test dialog of the Y Axis, only 1 motor is running

SebKuzminsky, 2018-06-22:
I confirm that the "Test this axis" dialog of Stepgen (in 2.7.14) only sends the step & dir signals to the first set of parport pins.

The final generated ini config file has the correct nets and sends the step & dir signals to all selected parport pins.

This is a bug in stepconf, patches welcome.

SebKuzminsky, 2018-06-22:
As a workaround I suggest connecting both Y drivers directly to the same parport pins (4 & 5 in your case), then telling stepconf you only have one Y motor.

andypugh, 2022-11-13:
Should we attempt to fix this, or simply document the axis test as a single motor test?

hansu, 2022-11-13:
Hmm it seems a bit uncomfortable to change the wiring for testing motors. Can't be that big change ...

andypugh, 2022-11-13:
I had a look, It's not a trivial fix, I think it would require a different HAL fragment to be generated. 
https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/usr_intf/stepconf/stepconf.py#L1483