LinuxCNC/linuxcnc #455: Editing an O-word sub while lInuxCNC is live gives a spurious "Duplicate O-word error"

andypugh, 2018-07-04:
Discussion and details here:
https://forum.linuxcnc.org/20-g-code/27728-duplicate-o-word-label-error-in-version-2-6?limitstart=0#113181

KurtJacobson, 2018-07-26:
I can confirm this on the latest master. It only happens on the first call to the sub after editing, subsequent calls behave as expected.

andypugh, 2019-01-25:
Do we think this is a big enough problem to leave as an open issue? 
Maybe we just document the behaviour?

zultron, 2019-02-01:
Heh, I hadn't seen this one.

This does indeed have to do with caching; see the `offset_map` in `src/emc/rs274ngc`.  I'm submitting a PR for it in #555.

manymachines, 2019-06-09:
I verified this fix with the current 2.8 branch (I'm the guy from the forum discussion).  I think this issue can be closed now?  Thanks!

closed 2019-06-09 by andypugh