LinuxCNC/linuxcnc #461: Gantry racks if keyboard arrow keys are pressed while homing in master branch

rodw-au, 2018-07-09:
Applies to master branch.

User on the forum reported that when homing a joint axis if he pressed the keyboard arrow keys, the gantry would rack.
https://forum.linuxcnc.org/49-basic-configuration/34819-help-required-2-8-slaved-axis?start=10#113859
Behaviour is confirmed on my gantry machine.   
## Here are the steps I follow to reproduce the issue:

 1. Commence home all on a joint axis (gantry) machine
 2. Press left or right arrow key on keyboard before homing is completed.
 
## This is what I expected to happen:
Keyboard input of jog buttons should be disabled when homing

## This is what happened instead:
Appeared to drop back into joint mode, one joint stopped moving and the other side kept moving. Gantry was racked but I think motion was stopped after error message(s) was displayed.

## Information about my hardware and software:

 * I am using this Linux distribution: Linux Mint
4.4.32-rt43 #1 SMP PREEMPT RT Wed Dec 7 23:00:11 AEST 2016 x86_64 x86_64 x86_64 GNU/Linux

 * I am running ...
   * A binary I built myself - External Offsets branch Commit 5acf4d1  
    https://github.com/LinuxCNC/linuxcnc/tree/dgarr/external_offsets
   *User was using Master branch, not sure of commit or version
 * I am using this user interface (GUI): AXIS
 * I am using this interface hardware vendor and chipset: Mesa 7i76e

rodw-au, 2018-07-09:
Screenshot after motion stopped
[image]

KurtJacobson, 2018-07-26:
@rodw-au can you confirm if 8b8e44ee89ac1da0e28cba848f892a5149ecda8e fixes this issue?

c-morley, 2019-01-10:
@rodw-au Is this issue fixed?

rodw-au, 2019-01-11:
Sorry, I have not tested it. I did not want to overwrite my external offsets branch config as the plasma component has been removed from the branch before committing to master. Dewey said in the email list that he would move it to another branch but I could not ever see where he'd done that.

c-morley, 2019-01-11:
Ok thanks I'll leave this open for now then.