LinuxCNC/linuxcnc #470: Missing axis INI setting with trivkins should default to joint settings or cause a warning

c-morley, 2018-07-21:
Missing [AXIS} setting such as MAX_VELOCITY and MAX_ACCELERATION default to a probably hard coded number such as 1unit per second for velocity.
This makes trouble-shooting non obvious for things such as max-jog-velocity GUI settings not being honored.
This really should either fail or create a warning or default to the relevant joint's setting.

Relevant for master because of joint/axis work.

rene-dev, 2018-08-03:
IMHO there should be no axis section on trivkins machines, it should just copy the joint values.

andypugh, 2018-08-03:
On 3 August 2018 at 20:53, Rene Hopf <notifications@github.com> wrote:

> IMHO there should be no axis section on trivkins machines, it should just
> copy the joint values.
>
Or there should be no joint section....

The wizards can cope, anyone writing an INI by hand can cope, it's more
generic as it is, although it is inelegant.

It does make it very easy to make a trivkins into a gantry if the joints
and axes remain seperated.

-- 
atp

c-morley, 2018-08-04:
I agree with either - but no axis section probably would be easier since it has less in it.
I would rather get an error code then use default numbers.
If in a non-triv config one should get an error if either section is missing.

Being able to switch a trivkins to a gantry with out doing some INI work seems unimportant compared with getting wrong a relatively hard to track down behavior

andypugh, 2018-08-04:
I really don't think this is an issue. Even I don't write an INI from scratch, and anyone who does can reasonably be expected to know what it takes. Or they wouldn't be trying.

c-morley, 2018-08-04:
Well I have worked on linuxcnc for a very long time.
I think I have more then the average newbe's knowledge.
I was caught by this.

gmoccapy, 2018-08-04:
IMHO any error in an config file should lead to an error. So also in this case!

c-morley, 2018-11-07:
Looks like in master @dngarrett 's commit 60d78df cover this now - I didn't notice it got added.
Thanks Dewey!

closed 2018-11-07 by c-morley