LinuxCNC/linuxcnc #491: Parker AC10 VFD mb2hal-configuration

chaotix-, 2018-08-23:
I connected my Parker AC10 VFD to LinuxCNC via mb2hal recently.

Is there interest in documenting what was necessary to get all the pieces working together? It would consist of the necessary settings for the VFD, mb2hal.ini, HAL configuration and a PYVCP-XML-file (to show current speed and at-speed-indicator LED).

Should I create a pull request and if so where should it go? At src/hal/user_comps/mb2hal/examples or docs/src/???

Or should I just put it in the issue here and someone will included it?

superfunk58, 2018-09-02:
YES, there is GREAT interest in getting a little walkthrough - i am currently trying to control a parker ac10 with linuxcnc via modbus as well...

superfunk58, 2018-09-04:
> Here is just a quick summary:
>
> You need LinuxCNC > 2.7.14 (e.g. the next release), the master branch or need to compile mb2hal yourself with the PR #487 and #489 .
>
> Configure the AC10 so your spindle is working. Then set F203 to 10. Then you can control it via ModBus. Use register 0x2000 to start/stop and set direction, register 0x010d (F113) to set the speed and register 0x1000+ to read the current status (see manual for more information).
>
> You can use the attached configuration files to get you started. They are more compex than is necessary as I tried to display the current and mean power of the spindle. But I don't think the values retrieved from the VFD are correct and are much too high. So you can strip out the power calculation which should shrink the HAL configuration in half.
>
> Please be aware that the minimum and maximum frequencies for the spindle need to be set in the AC10 configuration registers (F111 and F112) and the custom.hal file. And of course I don't give any guarantees that this won't burn down you house or cause an earthquake. All I can say is that it works fine for me.
>
> Github does not allow me to attach the .ini and .hal files so you need to copy & paste them.
>
> I hope, it helps.
>
> mb2hal.ini
>
> [MB2HAL_INIT]
>
> #OPTIONAL: Debug level of init and INI file parsing.
> # 0 = silent.
> # 1 = error messages (default).
> # 2 = OK confirmation messages.
> # 3 = debugging messages.
>
> INIT_DEBUG=3
> [... 312 lines cut ...]
>
> net vfd-power-mean pyvcp.vfd-power-mean
> net vfd-power pyvcp.vfd-power-cur
> net spindle-at-speed pyvcp.spindle-at-speed
> net spindle-speed-in pyvcp.spindle-speed-in

thanks a lot for all the detailed info,
i will definitely give it a go ....

andypugh, 2018-10-10:
Maybe this belongs on the Wiki? Leaving it here as an open issue looks bad...

chaotix-, 2018-10-14:
I have added a wiki page at http://wiki.linuxcnc.org/cgi-bin/wiki.pl?VFD_Parker_AC10.

closed 2018-10-14 by chaotix-