LinuxCNC/linuxcnc #503: New multi spindle bugs to fix in pncconf

c-morley, 2018-10-06:
In the work for multi spindles there were name changes besides adding .0 to HAL pins.
eg
halui.spindle-override.count-enable (note hyphen between spindle and override)
vrs
halui.spindle.0.override.count-enable 

One might have expected:
halui.spindle-override.0.count-enable 

Pncconf was not updated for these new pin names, before i fix them I  am confirming that the halui pin names are what you wanted.

c-morley, 2018-10-06:
I will just add:
While halui.spindle.0.override.... etc is consistent with the new spindle pins,
it is not consistent with the other override pins such as:
haui.feed-override...etc

andypugh, 2018-10-06:
Yes, the pin names are as-intended. 
The naming is changed to be hierarchical. 
It is the override for spindle 0, not the zeroth override. 

Component.item.N.function.pin

(

c-morley, 2018-10-06:
ok thanks I'll fix pncconf.

closed 2018-10-06 by c-morley

KurtJacobson, 2018-10-06:
@andypugh I like that, makes the overrides sort with the respective spindle in a tree view.