LinuxCNC/linuxcnc #514: pncconf error with units in fields

TurBoss, 2018-10-23:
I'm having issues with unit fields in pncconf

scale set to 160000 instead of 160 when click accept/ok in axis calculator
and travel area multiplied by 10 when click on "back" button

## Information about my hardware and software:

 * I am using this Linuxcnc stretch iso:
 * I am running ...
   * [X] A binary version from linuxcnc.org buildbot.linuxcnc.org

> hazzy:
TurBoss: I was able to reproduce the pncconf problem. It occurs when a comma is used as the decimal seperator
It appears that the comma is ignored, so 120,0 is converted to the number 1200
TurBoss
hazzy: but I didn't put any dot or comma
it does automatically
hazzy
Yes, GTK puts it in automaticaly because of your local
That also explains why your STEP_SCALE was 1000 times what it should be the other day. The calculator adds three decimal places, and GTK correctly used a comma for the decimal seperator because of your local, but PNCConf interpreted that as 160000 instead of 160

Thanks

TurBoss, 2018-10-23:
linux 4.9.0-8-rt-amd64
LINUXCNC - 2.8.0-pre1-3928-g0cab36566

c-morley, 2018-10-23:
What local are you using?

TurBoss, 2018-10-23:
my system language?
spanish

c-morley, 2018-10-23:
Thanks, I'l try to look into it soon.

TurBoss, 2018-11-17:
Hello, 

any news about this?
Thanks

c-morley, 2018-11-18:
Sorry I got busy...I have confirmed the error.
A work around would be to run pncconf in an English locale.
running from terminal like this:
LANG=C pncconf

c-morley, 2018-11-18:
Ok I pushed a fix for this to master.
Let me know if this cures the problem for you.

TurBoss, 2018-11-19:
no updates on buildbot

SebKuzminsky, 2018-11-19:
There was a spurious build failure on the master branch, which prevented the buildbot from producing packages with this fix.  I just kicked off the build again, hopefully it'll pass and the debs should be ready in a few hours.

TurBoss, 2018-11-20:
fixed now shows 125,0000

Thanks
closing this

closed 2018-11-20 by TurBoss