LinuxCNC/linuxcnc #523: angular only movement doesn't honour max velocity setting

c-morley, 2018-11-03:
## Here are the steps I follow to reproduce the issue:

home AXIS's 9 axis sample config
Set the max Velocity slider to 0
in MDI enter: g0 A20

## This is what I expected to happen:

linuxcnc to wait till max Velocity slider is above 0

## This is what happened instead:

Linuxcnc moves A axis to 0 

## It worked properly before this:

Not sure but 2.7.14 has the same problem

## More info:

Angular movement does honour rapid override setting.
It honours max Velocity setting if another linear axis is included.

rene-dev, 2018-11-03:
using the same max value for angular and linear moves does not make sense. So really, you need 2 sliders, for max linear and angular velocity?

c-morley, 2018-11-03:
Max velocity was a hack to create a way to slow down rapids a long time ago.
Now we have a proper rapid overrride for this.

But it's a surprise that the max velocity didn't limit angular velocity,

Thinking out loud:
if there is linear : limit base on max linear velocity.
if there is angular only limit based on max angular velocity.

So really to make a slider makes sense it should be in percentage.

Chris M

rene-dev, 2018-11-03:
ah, yes. I like the idea of a factor. but of what? fastest linear/angular axis?

c-morley, 2018-11-03:
Thinking out loud again:
if there is linear : limit base on percentage of max linear velocity.
if there is angular only limit based percentage of max angular velocity.

basically the same as feed and rapid overrides,
Or else just get rid of it all together ...

zultron, 2018-11-06:
I have a fix from another branch for this, but I haven't tested it with Axis or the other UIs.

If someone wants to try it out, I submitted it in PR #525.

zultron, 2018-11-06:
I [raised this issue][1] on emc-developers back in 2017-01, asking for input about how to treat the slider value, as a linear velocity or as a percentage.  Summarizing the feedback:

- @cradek pointed out that the touchy GUI needs a linear velocity for continuous jogs
- @samcoinc likes to know the velocity limit in linear units, not just a percent scale, when adjusting the slider in the GUI

These issues are easily addressed in this implementation.  Since the limits ended up being separate values for linear and angular velocity, touchy jogging will be unaffected.  In `axis.py`, the angular velocity is simply scaled by the computed percentage of the slider value, and the result is sent to motion, so the appearance of the GUI is unchanged.

Also, at the time, @gmoccapy volunteered to update the gmoccapy GUI.  I hope he's still agreeable to do so!

[1]: https://sourceforge.net/p/emc/mailman/emc-developers/thread/34842cc2-93de-efcb-98a6-676853e8cf05%40zultron.com/#msg35589852

c-morley, 2018-11-07:
Sorry i answered on the pr #525 - but here is a copy:

I tested John's fix in simulation and it indeed limits angular only movement - Yaay!
Thanks for addressing this John.
The units on the slider of course, make no sense for angular movement.

reading John's comments here:

The .ini docs place MAX_LINEAR_VELOCITY and
MAX_ANGULAR_VELOCITY in the [DISPLAY] section; the former is the
maximum possible value on the max velocity slider. The MAX_VELOCITY
param is in the [TRAJ] section, and is the set value on the slider.
This naming makes it hard to add the missing angular equivalent to
[TRAJ]MAX_VELOCITY

I looked at the docs for master and those terms in the TRAJ are different.
eg. MAX_LINEAR_VELOCITY -for master
I haven't dug into the code to see what actually uses these entries and how they should be spelled- I wonder if the INI docs lie.... I know in GUI building the TRAJ and DISPLAY terms were often confused.

For the GUI I wonder why you would limit the slider at all - it should be the machine's max possible velocity - which I assume would be from the TRAJ or JOINT sections.

overall thought (not all are directly about the patch):

This is a excellent improvement.
Should we put this in 2.7 ?- while it is a bug fix, it touches a lot of code.
There is question what the slider units should display
If not %, there is question where the GUI should read it's max setting from
I wonder why we need this capability anymore - overrides seem to do it better.

gmoccapy, 2018-11-11:
IMHO we should favorite a two slider configuration.
I would include than a slider with percentage indication in gmoccapy and that one would control both velocities. May be others want two slider, they would be able to do that with embedded tabs.

Just tell me when I need to do work on gmoccapy.

Norbert