LinuxCNC/linuxcnc #524: reverse-run-master2 acceleration issue

phillc54, 2018-11-04:
## Here are the steps I follow to reproduce the issue:

 1. Clone linuxcnc repo, checkout reverse-run-master2 branch, build lcnc and then run with axis_mm from sim/axis
 2. Load this code:
;line test
G21
G64 P0.005
M52 P1
F1000
G0 X10 Y10
M3 S1
G1 X90
G1 X100 Y20
M5
G0 X0 Y0
M2

 3. While running, do setp motion.adaptive-feed -1 until the control point reaches the start point then do setp motion.adaptive-feed 1, motion restarts in the forward direction but acceleration is painfully slow.
If I only do a partial reverse then acceleration appears normal (until a full reverse is applied).
If I remove line 9 (G1 X100 Y20) acceleration behaves normally.
If I use arcs, acceleration also appears normal.

## Information about my hardware and software:

 * I am using Debian GNU/Linux 9.4 (stretch)
 * I am using Linux stretch 4.9.0-6-rt-amd64 #1 SMP PREEMPT RT Debian 4.9.88-1+deb9u1 (2018-05-07) x86_64 GNU/Linux
 * I am running a rip version
 * I am using this LinuxCNC version, v2.8.0~pre1~feature-reverse-run-master2~bef8d9d52
 * I am using AXIS
 * I am using axis_mm sim from the configuration selector

phillc54, 2018-11-06:
I should have added that slow acceleration is only for the first segment, when the next segment is started then acceleration returns to normal.

bhushan00, 2019-05-29:
i am using 2.8 ~pre version but reverse run function isnt working on that

phillc54, 2019-05-29:
feature/reverse-run-master2 has not worked for quite some time as it has fallen behind other updates.
There is an up to date branch here that is working for some folk https://github.com/phillc54/linuxcnc/tree/reverse-run

robEllenberg, 2019-06-14:
@phillc54 Do you still see this behavior in master after Andy's merge?

phillc54, 2019-06-14:
@robEllenberg 62d4c6e seemed to have fixed it

closed 2019-06-14 by robEllenberg