LinuxCNC/linuxcnc #546: Acceleration violations in linuxCNC 2.7 and 2.8

pcw-mesa, 2018-12-30:
Both current versions of LinuxCNC have acceleration constraint violations in some circumstances
While testing a forum posters gcode files ( https://forum.linuxcnc.org/9-installing-linuxcnc/35744-not-a-bug-report ) i found that I could get acceleration violations with one of the test files: \

http://freeby.mesanet.com/1.ngc

I have aggressive velocity and acceleration limits on my test configuration: 20 IPS and 200 IPS^2
and that may play a part, but this error is reproducible in both  2.7 and 2.8 (first error around line 920)

http://freeby.mesanet.com/28ferror.png
http://freeby.mesanet.com/28ferrorb.png

The second plot has the ferror limits widened enough so the following error fault is not generated

pcw-mesa, 2019-01-01:
I discovered that AFAICT this bug is triggered by very small G64 PXXX values
The example files provided on the forum had G64 P0.001 in mm 
if this is set to .003 or greater the acceleration constraint is not violated

samcoinc, 2019-01-01:
I think we have ran across that bug before...

pcw-mesa, 2019-01-02:
Yes I remember. This (tiny G64 Pxxx values) seem to mainly effect configs with high velocity/acceleration. It looks like the forum posters issue is not the same, I can duplicate it here with a mm LinuxCNC 2.8 config:

http://freeby.mesanet.com/fe28.png

Axis A (Joint3) has a maximum velocity of 200 degrees per second and acceleration of 2000 degrees
per second^2 you can see that it violates the acceleration constraint by about 50% and the violation is not dependent on PXXX values ( Interestingly, it _does_ go away with G61 however )

robEllenberg, 2019-01-05:
Thanks for providing the scope traces and test program. It was possible to slice out a short (20 line) portion of it to reproduce the issue in sim. Would you mind giving my fix branch a try to see if the fix works on your platform?

pcw-mesa, 2019-01-05:
Sure, how do I get you patches? bear with me, I have a terrible flu so keep it simple :-)

robEllenberg, 2019-01-07:
You can get the branch from my github mirror of linuxcnc here:

https://github.com/robEllenberg/linuxcnc-mirror

the branch name is *bug546-tangent-segment-acceleration-corner-cases*

pcw-mesa, 2019-01-07:
OK tested that branch, I does seem to fix the "small G64PXXX value" problem
but I still get the same acceleration error (looks like acceleration exceeds constraint by 50%)
in the 1_wrap.ngc file with my configuration:

http://freeby.mesanet.com/a-err.png

(bottom trace is acceleration, ini file constraint is 2000 but acceleration reaches ~3000)

robEllenberg, 2019-01-07:
Is that at the end of the program? It looks like it's coming up to the last segment too fast; possible an issue with the optimizer. Is 1_wrap.ngc different from 1.ngc linked above? If so, could you post a link to it? I'll try to reproduce it with the debug output on and see where it's failing.

robEllenberg, 2019-01-08:
Thanks! I was able to reproduce it by doing a find / replace of "Y" with "A" in the original file as well. I just pushed a fix to that same branch (currently running in sim locally without violations so far). There was a parabolic blending case where it wasn't reducing acceleration correctly, causing the violation you saw.

pcw-mesa, 2019-01-08:
Yay! and thanks! both bugs fixed AFAICT

SebKuzminsky, 2023-04-23:
We believe this is fixed in 2.7 and newer: https://github.com/LinuxCNC/linuxcnc/commit/04d7be913e57dfebff9f2e293f8663603ef9c59c

PR #547 (2019-04-28): Bug546 tangent segment acceleration corner cases
Fixes #546 due to several corner cases in the optimization of approximately tangent segments. I've tested the fix in simulation against the test case, and the violations disappear (even for much tighter blend tolerances like 0.00001mm).
closed 2023-04-23 by SebKuzminsky