LinuxCNC/linuxcnc #565: Ignore comment lines during automatic update of machine config

manymachines, 2019-03-03:
## Here are the steps I follow to reproduce the issue:

 1.  Begin with  config [EMC]VERSION = 1.0
     The .hal file contains an out-of-date pin such as 'motion.spindle-on' in a comment line
     #net spindle-relay motion.spindle-on hm2_5i25.0.gpio.033.out

 2.  Start Linux CNC (current master)
 3.  Select 'Yes' in the 'Confirm automatic update' pop-up window (updates to [EMC] VERSION=1.1)

## This is what I expected to happen:

Comment lines in the .hal file would not be updated.

## This is what happened instead:

The .hal file comment line was updated from:
#net spindle-relay motion.spindle-on hm2_5i25.0.gpio.033.out
to
#net spindle-relay spindle.0.on hm2_5i25.0.gpio.033.out

It's a minor issue but say I had comments related to an older config version.  After the automatic update the comments may not make any sense.  I'd suggest not to update comment lines in the automatic update.

In my case I had manually updated my config to correct the old pin motion.spindle-on.  Then I later ran the automatic update and diffed the files and noticed the comment line was changed.

andypugh, 2019-03-03:
I can see two scenarios: 
1) The comments are documentation for the integrator, and in that case updating does not make sense
2) The commented-out line is an actual line of code that might be un-commented in the future, in that case making it into a functional line would make sense. 

ie, I can see arguments both ways and would prefer to see a few more opinions.

c-morley, 2019-03-04:
Would it be difficult to have the conversion program ask the user when the problem is detected?

Chris M

andypugh, 2019-03-04:
> Would it be difficult to have the conversion program ask the user when the problem is detected?

Yes. It would (probably) require two complete sets of regexes to handle both choices.

jethornton, 2019-03-07:
Can you do both? Leave the old comment and add the new line suitable to master?

andypugh, 2019-03-08:
jethornton: Currently it is just a big search-and-replace. Special handling for comments would be an awful lot more work. 

I could exclude comments fairly easily with a change to the regex, but I am really not sure that that makes sense. The original comments are backed-up in the archive folder anyway.

jethornton, 2019-03-08:
@andypugh  Probably the effort is not worth the result.
@manymachines can you close this?

manymachines, 2019-03-10:
I understand that the comments are intentionally modified to avoid a future error if the user were to un-comment the line.

closed 2019-03-10 by manymachines