LinuxCNC/linuxcnc #566: Tool Table Editor Error

jethornton, 2019-03-07:
Axis random tool changer simulation edit tool table fails in master.
## Here are the steps I follow to reproduce the issue:

 1. Run sim.axis radom_tc
 2. Click on File > Edit Tool Table

## This is what I expected to happen:
Tool Table Editor to appear.

## This is what happened instead:
Nothing.

## This is the error from the terminal:
Traceback (most recent call last):
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1540, in __call__
    return self.func(*args)
  File "/usr/bin/axis", line 2248, in edit_tooltable
    root_window.tk.call("exec", *e)
TclError: couldn't execute "xterm": no such file or directory

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`):
Tested on Linux Mint 18.3 and Debian 9
 * I am using this kernel version (shown by `uname -a`):
 * I am running ...
   * [X] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):
 * I am using this user interface AXIS:
 * I am using this interface hardware vendor and chipset simulator

TurBoss, 2019-03-07:
seems that LinuxMint is missing xterm
sudo apt update && sudo apt install xterm

jethornton, 2019-03-07:
fixed broken configs

closed 2019-03-07 by jethornton