LinuxCNC/linuxcnc #567: halcmd alias command deletes original pin/parameter

pcw-mesa, 2019-03-08:
Halcmds alias function deletes the original pin/parameter 
contrary to the man page for halcmd. 

This is true on LinuxCNC 2.7 and 2.8

To test
halcmd alias pin motion.adaptive-feed maf
halcmd show pin | grep adaptive 
 (motion.adaptive-feed is gone)

andypugh, 2019-03-08:
Curious. Do you think that this needs a fix to the docs, or should both pins exist?

pcw-mesa, 2019-03-09:
My opinion is that because that function is called "alias" it should not remove the original name
wonder what the author thinks...
(also removing names leads to some pretty weird situations where say signals are netted to pins
in the hal file that disappear... I suspect this is OK since I guess names just resolve to pointers but still weird...)

andypugh, 2019-03-09:
I tested part of that, if you alias a pin it remains in the same net with the new name.

andypugh, 2019-03-09:
This might just be an issue with "show pin". You can still use the original name in a net command. 
(and "show pin" shows the alias netted to the new signal)

pcw-mesa, 2019-03-09:
OK thanks for testing that, I guess this makes some kind sense of sense as a listing of 
pin --> signals connections would be a bit confusing with multiple names I think I will close this
with a note that this behavior could be better documented.

closed 2019-03-09 by pcw-mesa

andypugh, 2019-03-09:
8841b630976745a8c38a3b2318762c1c1e3fb783