LinuxCNC/linuxcnc #569: RS274NGC_STARTUP_CODE Multiple P Words

phillc54, 2019-03-17:
## Here are the steps I follow to reproduce the issue:
 1. edit sim/axis/axis.ini
 2. in the [RS274NGC] section
 3. add RS274NGC_STARTUP_CODE = G64P0.1 M52P1
 4. save then run

## This is what I expected to happen:
LinuxCNC to start with both above commands in effect

## This is what happened instead:
LinuxCNC stopped with the following errors on screen:
emc/task/emctask.cc 405: interp_error: Multiple p words on one line
Multiple p words on one line
emc/task/emctaskmain.cc 3085: can't initialize interpreter

## Information about my hardware and software:
 * I am using Debian GNU/Linux 9.8 (stretch)
 * I am using SMP PREEMPT RT Debian 4.9.88-1+deb9u1 (2018-05-07)
 * I am running a binary version from linuxcnc.org (including buildbot.linuxcnc.org)
 * I am using LinuxCNC v2.8.0-pre1-4634-g8d71cbc22
 * I am using AXIS
 * I am using sim only

andypugh, 2019-03-17:
Keyword order on a line has no meaning in the G-code specs. Commands are interpreted in order of precedence and modifiers can appear anywhere on the line. 
So G64 P0.1 M52 could mean "G64" then "P0.1 M52"
The error message is correct, you are not allowed to use the same letter command twice on the same line.
One workaround is to use
RS274NGC_STARTUP_CODE = O<startup> CALL
and to put the startup G-codes on separate lines in an O<startup>SUB structure in a file called startup.ngc

phillc54, 2019-03-18:
Thanks for the explanation, I assumed that the startup codes would be acted on sequentially rather than as a single line.

c-morley, 2019-03-18:
I agree with Phillip here. I can't imagine it would be difficult to have linuxcnc apply them sequentially. Is there a reason not too?

Chris M

andypugh, 2019-03-18:
On Mon, 18 Mar 2019 at 01:20, c-morley <notifications@github.com> wrote:

> I agree with Phillip here. I can't imagine it would be difficult to have
> linuxcnc apply them sequentially. Is there a reason not too?

Yes. That isn't how G-code works. So you would have to re-write the
interpreter and break the G-code standards at the same time.

However, I think it would be fairly easy to accept a comma-delimited list.
I am looking at that possibility now.

-- 
atp

c-morley, 2019-03-18:
I wasnt suggesting changing the code is interpreted,  only how the code is pulled from the INI file and entered into the interpreter.

Chris

andypugh, 2019-03-18:
We could accept several STARTUP_CODE lines, like MDI_COMMAND and some others. But a list is probably easier.

phillc54, 2019-03-18:
My original thinking was the same as Chris's, that this should only be for how it is read in from the ini file.

zultron, 2019-03-18:
The fundamental block-reading routines in interp only allow unique flags within one block, and repeats are errors, as @phillc54 found out.

Some of the options can be repeated, but this one apparently can't.  It doesn't look too hard, following other examples like `REMAP` in `emc/rs274ngc/rs274ngc_pre.cc` for `emc/task/emctaskmain.cc` and using `inifile.findall()` instead of `inifile.find()` in `emc/usr_intf/axis/scripts/axis.py`.  But given how easy this is, there couldn't have been much past demand for it or we'd have it already.

So, +1 on @andypugh's suggestion.  It works today; see the example in `tests/motion-logger/startup-gcode-abort/test.ini`.

phillc54, 2019-03-18:
I am happy to keep it as it is seeing as how the o call works.
Maybe just add a note to the docs [here ](http://linuxcnc.org/docs/devel/html/config/ini-config.html#_rs274ngc_section)

andypugh, 2019-03-18:
I have implemented a comma-separated list, it is a small change to emctask.cc

  if (retval > INTERP_MIN_ERROR) {  // I'd think this should be fatal.
	print_interp_error(retval);
    } else {
	if (0 != rs274ngc_startup_code[0]) {
        **char startupcmd[LINELEN];
        char* ptr;
        strcpy(startupcmd, rs274ngc_startup_code);
        ptr = strtok(startupcmd, ",");**

        **while (ptr != NULL){**
            retval = interp.execute(**ptr**);
            while (retval == INTERP_EXECUTE_FINISH) {
                retval = interp.execute(0);
            }
            if (retval > INTERP_MIN_ERROR) {
                print_interp_error(retval);
            }
            **ptr = strtok(NULL, ",");**
	    }
	}
    }

Which works, but Axis complains about the comma, and I suspect that gremlin and Gmoccapy might too, as they all also take a look at the RS274NGC_STARTUP_CODE entry.

andypugh, 2019-03-18:
Also, I prefer the idea of multiple lines and using ini.findall()

closed 2019-03-20 by phillc54