LinuxCNC/linuxcnc #570: Undefined reference to negate inline function

AndreaRicchi, 2019-03-20:
## Here are the steps I follow to reproduce the issue:

 1. create a project using linuxcnc library;
 2. try to compile;

## This is what I expected to happen:
Compile without errors.

## This is what happened instead:
Compile with linker errors: undefined reference to function negate, file blendmath.c, line 85.

## Information about my hardware and software:

 * I am using this Linux Mint version 19.1;
 * I am running a simple Qt application built with G++ 7.3.0;
 * I am using the latest LinuxCNC version;

robEllenberg, 2019-06-14:
@AndreaRicchi do you see this issue in 2.7 as well? If not, with your fix merged, can we consider this issue fixed?

AndreaRicchi, 2019-06-18:
@robEllenberg I confirm that the issue is fixed. Thanks

closed 2019-06-19 by robEllenberg