LinuxCNC/linuxcnc #572: "run from here" causes multiple calls to manual toolchange

tlight9, 2019-03-24:
A code snip from the program i used to produce this:
%
(Mill_holder)
N1 G28 Z0
N2 T9 M06 (0.25 .250 END MILL) 
N3 S2500 M03
N4 G54 G17 G00 G90  X-2.319 Y-0.1925 
N5 G43 Z-0.3 H09
N6 G00
N7 Z-0.3625
N8 G01 Z-0.4625 F1.5
N82 G28 Z0
N83 T32 M06 (0.595 3/4" THREAD MILL) 
N84 S2500 M03
N85 G17 G00 G90  X-3.0 Y-0.1925 
N86 G43 Z-0.3 H32
N87 G00
N88 Z-0.425
N89 G01 Z-0.4625 F3.0
N129 G28 Z0
N130 T1 M06 (.250 SPOT DRILL) 
N131 S2000 M03
N132 G17 G00 G90  X-3.0 Y-0.1925 
N133 G43 Z-0.3 H01
N134 G98 G81 Z-1.075 R-0.925 F3.2
N135 G80
N252 G28 Y0
N253 M30
%

## Here are the steps I follow to reproduce the issue:

 1. run from here at line N130 
 2.
 3.

## This is what I expected to happen:
Tool change request for tool 1

## This is what happened instead:
Tool change request for tool 9
Tool change request for tool 32
then Tool change request for tool 1

## It worked properly in 2.7.14
After upgrading from LinuxCNC 2.7.14 to 2.8 master using synaptic package manager

 * I am using Debian GNU/Linux 7.11 (wheezy)
 * I am using this kernel Linux Wheezy 3.4-9-rtai-686-pae #1 SMP PREEMPT Debian 3.4.55-4linuxcnc i686 GNU/Linux
 * I am running ...
   * [X ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)

 * I am using LinuxCNC version 2.8 (master)
 * I am using AXIS
 * I am using a MESA 7I90 with a 7I47S daughter card

andypugh, 2019-03-26:
I can't replicate the behaviour on my machine.

tlight9, 2019-03-26:
Further investigation shows that commenting out:
REMAP=M6   modalgroup=6  prolog=change_prolog ngc=change epilog=change_epilog
in the config file clears up the problem. So the issue appears to be in remap. I didn't notice this behavior in 2.7.14

zultron, 2019-03-28:
Duplicate of #246