LinuxCNC/linuxcnc #575: Python Interface - stat.axis dict reports 0 velocity in auto

Eranziel, 2019-04-03:
As the title says - when running in auto mode or mdi mode, stat.axis has a velocity value of 0 for all axes. When jogging in manual mode, velocity value is correct. 

## Here are the steps I follow to reproduce the issue:
 1. Have a loop which prints out stat.axis data for each axis. For example:
`self.status.poll()`
`for i in range(3): print("Axis {}: {}".format(i, self.status.axis[i]))`
 2. Open a file and run it, or enter a command in MDI and run it.

## This is what I expected to happen:
Loop prints the axis dictionary for each axis, velocity entry should match the real-world velocity of that axis.

## This is what happened instead:
Velocity entry for all axes is 0 even when moving.

## It worked properly before this:
LinuxCNC 2.7.11 worked.

## Information about my hardware and software:
Tested on two machines. 1) prototype mill, 2) desktop with simulator only. 
 * I am using this Linux distribution and version: 1) Debian 9.6 (LinuxCNC image) with KDE desktop. 2) Kubuntu 16.04.6 LTS.
 * I am using this kernel version: 1) 4.9.0-8-rt-amd64 #1 SMP PREEMPT RT   2) 4.9.47-rt37 #1 SMP PREEMPT RT
 * I am running ...
   * 1) A binary version from buildbot.linuxcnc.org: v2.8.0-pre1-4106-g4440d31ff
   * 2) A binary I built myself from github v2.8.0-pre1-4628-g39ab67a.
 * I am using a custom interface. 
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): 1) Mesa 7i92e + 7i76. 2) None, used as simulator only.

andypugh, 2019-04-05:
This is probably a side-effect of the joints-axes changes. 
Does the joint dictionary report the correct numbers?

andypugh, 2019-04-05:
I have never quite worked out where the values get in to the Python interface. 
Some of the puzzle is here: https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/usr_intf/axis/extensions/emcmodule.cc#L583

Eranziel, 2019-04-05:
Just tested, and the joint dictionary does report the correct numbers.

Did some digging around in the file you linked, and it looks like the axis/joint data comes from the EMC_STAT class. See here: https://github.com/LinuxCNC/linuxcnc/blob/73f44f829cbfe7066eae5c191d5578cd8ce10628/src/emc/usr_intf/axis/extensions/emcmodule.cc#L84. I'll do some more digging and see if I can find where EMC_STAT gets the axis velocities from.

Possibly linked: I see that the HAL pins axis.*.teleop-vel-cmd also show 0 when moving in auto, but show the correct velocity when moving in manual.

Eranziel, 2019-04-05:
Maybe found it... someone let me know if I'm on the right track. You guys have a lot of code to dig through, haha. ;)

I think that the EMC stat axis data gets updated in this function: https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/task/taskintf.cc#L559

That value seems to come from control.c, where teleop_vel_cmd gets updated here: https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/motion/control.c#L1402, which then gets copied to the HAL component (https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/motion/control.c#L1915) and to EMC stat (https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/motion/control.c#L2094).

All that said, I'm not sure where in there things are going wrong.