LinuxCNC/linuxcnc #576: Wheezy has been EOL for 11 months. Why is the ISO still listed as current & available?

fake-name, 2019-04-04:
Wheezy left long-term support in **2018-05-31**. The package repos are no longer available and it's no longer receiving security updates.

Furthermore, the fact that repos are down means *you can't install packages anymore*, so even if you accept the security issues of running a now-dead OS, it's mostly useless anyways.

Please remove the unsafe ISOs, and maybe release a version that's still getting upstream support?

0wn3r, 2019-04-04:
You can change /etc/apt/sources.list to use Debian Archive to be able to install packages, but still you can't get security updates.
You should use:
**http://archive.debian.org/debian/** for the **wheezy main contrib non-free**
**wheezy-updates** and **security** are not available.

andypugh, 2019-04-04:
The workaround doesn't address the original point, though. 
We can't expect new users to know that they need to mess about with apt sources, and frankly they shouldn't need to. 
We should, at the very least, make the stretch ISO image the default. (And a Stretch RTAI image would also be useful).  
I think that one problem here is that not many people know how to create the ISO images.

mozmck, 2019-04-04:
While there are some other ISOs; as far as I know the Wheezy ISO is the only one with RTAI which makes it the only option for a lot of users (unless you install an even older OS).  As such it is fully useful for it's intended purpose.  Until someone can get a satisfactory RTAI kernel for a newer distribution, I think Wheezy will have to be the current ISO for anyone needing RTAI.
IMHO, EOL status for the base OS is not much concern for a machine control - although it would be nice if the sources.list were updated to the new package archive location.

I agree though that we should add any other available and tested ISO images to the download page.  I thought there was one for Jessie and Andy mentioned Stretch - but I don't see any mention of them on the website?

Eranziel, 2019-04-04:
What does RTAI provide or support that RT-PREEMPT does not? The only difference I've come across is that the Mesa ethernet-based FPGA boards require RT-PREEMPT, which to me is another reason to promote the Stretch ISO as the standard. 

The Stretch ISO is very well hidden, but I've not had any issues with it (other than disliking Xfce, which is merely personal preference). I saved a bookmark a while back to make life easier for myself, here's the link: [http://linuxcnc.org/testing-stretch-rtpreempt/](http://linuxcnc.org/testing-stretch-rtpreempt/)

andypugh, 2019-04-04:
Generally speaking the RTAI kernel performs better for software stepping (parallel port). 
Unfortunately that is where most newbies (myself included) started off so that does still seem like a good default.

andypugh, 2019-04-04:
Doing some poking about, the most recent RTAI download has patches for kernel 4.9.80
It seems that Debian Stretch is based on Kernel 4.9 so it seems like RTAI stretch is possible.

fake-name, 2019-04-04:
RTAI also supports 64 bit, so maybe it's finally time to join the 2000s?

jethornton, 2019-04-08:
Why can't we change /etc/apt/sources.list to use Debian Archive so the buildbot works and the docs get built?

andypugh, 2019-04-08:
Who has the keys to the buildbot? I know I don't.

SebKuzminsky, 2019-04-08:
I manage the buildbot, it runs on hardware in my house.  I just updated `/etc/apt/sources.list` on all the wheezy builders per @0wn3r 's comment above (and on the jessie builders which also needed a bump). 

I agree with @mozmck and @andypugh that people using software stepping via parport are better served by RTAI than they are by Preempt-RT, and that this is an important use-case to support well.  It would however be helpful to our users if we updated `/etc/apt/sources.list` on the Wheezy RTAI iso.

I spent a significant amount of time trying to get RTAI 5.0.1 to work with Linux 4.4 on Debian Jessie, and there were unexplained system hangs during realtime start & stop that both I and the RTAI dev team were unable to fix.  That's why we've never released it to users: we couldn't get it stable.  I haven't tried it for a long time, maybe things have gotten better since then.

@jepler did all the work on the Stretch Preempt-RT iso, and I believe it's mostly ready to go, we just never pushed that effort all the way across the finish line.

jepler, 2019-04-08:
The code to build the stretch iso images, including the helper scripts related to uploading the .iso and .zsync files to wlo, are all on github.

Someone will probably want to re-roll this to get current debian updates (including kernel security updates) and the current linuxcnc release.

I'm not likely to be spending time on it, so I encourage someone else to pick it up and do so.  I can arrange for github permissions to the repository, if nobody else is handy to do so.  Everything's here:  https://github.com/LinuxCNC/stretch-live-build

Permssion to upload to wlo is more complicated, someone who already has authorization needs to drop an ssh public key in the related shell account's authorized_keys.  Besides me, this list should include at least cradek and seb.

sebastiankuzminsky, 2019-04-08:
If someone picks up the building & testing of this updated Stretch ISO, I'll help with getting it up on wlo.

SebKuzminsky, 2019-04-08:
If someone picks up the building & testing of this updated Stretch ISO, I'll help with getting it up on wlo.

jethornton, 2019-04-09:
I'll give it a try

rene-dev, 2019-04-09:
I came across this PR the other day, jepler already started to update the docs: https://github.com/LinuxCNC/linuxcnc/pull/500

jethornton, 2019-04-09:
I cloned stretch-live-build and installed live-wrapper and ran do.sh but I get this error:
`john@deb9-fresh:~/stretch-live-build$ ./do.sh
Usage: -c [options]

-c: error: no such option: --customise
`
Any clue what went wrong?

jethornton, 2019-04-11:
I got it working but when I realized it was downloading a bunch of data and had used up 1.5GB of my monthly allotment I had to shut it down. If it takes more than 3 hours to build I can't build it as free data ends for me at 8am.

rene-dev, 2019-04-15:
While working on jessie, keep in mind that It will be EOL in june 2020.
I dont understand why we bother with the past, and EOL operating systems.
It would make much more sense to spend time to release the stretch iso. Stretch works great.

JonnyTech, 2019-05-11:
The last stretch images are from 2018, is anything more recent available?

andypugh, 2019-05-11:
The current 2.7.14 release was June 2018, so the Stretch ISO is up-to-date with the LinuxCNC release. 
I am not sure what is delaying the 2.8 release.

treki1, 2019-05-12:
Original:
Wie sieht es denn nun aus? Wann wird 2.8 und Stretch offiziell?

Translator:
How does it look now? When will 2.8 and Stretch become official?

andypugh, 2019-05-23:
The 2.6 Wheezy ISO file has been replaced with one that links to the archive repositories, so package installs etc will now work again. 
This is obviously a stogap, but work is under way to switch to the Stretch preempt-rt ISO as the default download.

closed 2019-05-23 by andypugh