LinuxCNC/linuxcnc #582: INI parameter access in G-Code

Eranziel, 2019-05-06:
## Here are the steps I follow to reproduce the issue:

 1. Create a .ngc file which reads an INI file parameter using `#<_ini[section]name>`.
 2. Try to run the file.

Here's a simple example:
```
#<xhome> = #<_ini[axis_x]min_limit>
g0 #<xhome>

M2
```

## This is what I expected to happen:
`#<_ini[axis_x]min_limit>` should return the value from the INI file. In this case, 0.0.

## This is what happened instead:
Interpreter gives the following error:
```
G-Code error in temp.ngc
Near line 2 of
/home/user/linuxcnc/nc_files/temp.ngc
Named parameter #<_ini[axis_x]min_limit> not defined
```

This functionality is mentioned in the [devel docs](http://linuxcnc.org/docs/devel/html/remap/remap.html#remap:referto-inifile-variables). The note at the bottom of the section implies it was part of a branch - either the functionality should be merged into master (my personal preference), or the documentation removed.

andypugh, 2019-05-06:
Have you enabled the feature using the FEATURE_MASK described in section 18?

http://linuxcnc.org/docs/devel/html/remap/remap.html#_optional_interpreter_features_ini_file_configuration

zultron, 2019-05-07:
The confusing comment @Eranziel refers to is also in the [2.7 documentation][1]:

> Caution | this section doesn’t really belong here but since it comes with the same branch, here it rests for now until its clear this will be merged. It should go into the gcode/overview Named Parameters section.

It looks like something that wasn't caught during review.

It would make good sense to explicitly note this is an experimental feature and add a pointer to section 18 for how to enable it.

[1]: http://linuxcnc.org/docs/2.7/html/remap/remap.html#remap:referto-inifile-variables

andypugh, 2019-05-07:
I can also see little reason for it not to default to “on” 
I am more wary of doing that with HAL pin values.

sebastiankuzminsky, 2019-05-07:
FWIW, I agree with @andypugh .  As far as possible, let's try to have a single set of features, always on.

SebKuzminsky, 2019-05-07:
FWIW I agree with @andypugh.  As far as possible let's try to have a single set of features, always on.

Eranziel, 2019-05-07:
Well, I completely missed that when I read section 18, my mistake.

However, I've now added `FEATURES = 8` to the RS274NGC section and I get the same result. Can anyone else confirm?

FWIW, my personal preference would be to default that feature to ON as well. I may or may not be biased by currently working on a project where I want to use it...

andypugh, 2019-05-07:
It's 4 to turn on INI parameter access, and 4 to turn on HAL pin access. 
(or 12 to make sure and turn on both)

Eranziel, 2019-05-07:
Total reading comprehension fail there, my bad. It does work when you turn it on. ;)

Shall I close this, or leave it open until the documentation gets adjusted?

andypugh, 2019-05-07:
Leave it open, as we seem to have decided to change something.

gmoccapy, 2019-08-07:
I agree with Andy, FEATURES = 12 should be the default value.

Norbert

phillc54, 2019-08-07:
FEATURES = 12 as the default makes sense to me

mozmck, 2019-08-09:
I vote for FEATURES = 12 as the default as well.  I've been using that for a long time for both INI settings and HAL pins without any problems.

andypugh, 2021-06-21:
We don't use "FEATURES" at all any more.
 74fc4491014b9de4141c44f58e391960e681343b

closed 2021-06-21 by andypugh