LinuxCNC/linuxcnc #592: after upgrading to 2.8 gmoccapy gives error then exit

Berkdan, 2019-05-16:
Hi Norbert,

I hope this is the right place to right my problem.

I installed and used gmoccapy with linuxcnc 2.7.14 without problem. I also installed glade 3.8.6 and followed your instructions to edit gmoccapy.glade file with glade 2 all was good so far but my problem started when i upgraded to linuxcnc 2.8 via synaptic from buildbot repo.
now, gmocccapy does not work. It gives me these error what ever i do to make it work. I hope you can help me.
I even built linuxcnc master from source then started as rip but no luck.
Thanks
Berkdan

The error message :

Found an error!
The following information may be useful in troubleshooting:

Traceback (most recent call last):
File "/usr/lib/python2.7/dist-packages/gladevcp/tooledit_widget.py", line 103, in __init__
self.set_col_visible(list='spyabcuvwdijq', bool= False, tab= '2')
File "/usr/lib/python2.7/dist-packages/gladevcp/tooledit_widget.py", line 401, in set_col_visible
renderer.set_property('visible', bool)
AttributeError: 'NoneType' object has no attribute 'set_property'

it gives me this error first when i press OK Gmoccapy screen appear below an error message again then when i click ok it shuts down.

the next error message is :

ound an error!
The following information may be useful in troubleshooting:

Found an error!
The following information may be useful in troubleshooting:

Traceback (most recent call last):
File "/usr/bin/gmoccapy", line 4755, in <module>
app = gmoccapy(sys.argv)
File "/usr/bin/gmoccapy", line 303, in __init__
self._init_tooleditor()
File "/usr/bin/gmoccapy", line 1055, in _init_tooleditor
self.widgets.tooledit1.set_filename(toolfile)
File "/usr/lib/python2.7/dist-packages/gladevcp/tooledit_widget.py", line 233, in set_filename
self.reload(None)
File "/usr/lib/python2.7/dist-packages/gladevcp/tooledit_widget.py", line 240, in reload
self.model.clear()
AttributeError: 'ToolEdit' object has no attribute 'model'

gmoccapy, 2019-08-07:
It seems, that your config do miss a valid tool file. 

Have you checked the terminal output for further information?
Please start a one of the supplied sim configs and report if they do work.

What release number does gmoccapy report? It should be 3.0.7.1

Norbert

gmoccapy, 2019-08-12:
Not able to reproduce that . tested on several pc and os, so i close this one

closed 2019-08-12 by gmoccapy