LinuxCNC/linuxcnc #595: AXIS won't home Z axis on lathe in 2.8

c-morley, 2019-05-22:
## Here are the steps I follow to reproduce the issue:
In my case the config has indiviual axis homing.
 1. home X (finishes)
 2.Home Z

Apparently someone else is having the same problem:
https://forum.linuxcnc.org/38-general-linuxcnc-questions/36636-lathe-z-axis-joint-not-homing-after-updating-to-2-8#134615

## This is what I expected to happen:
Z axis to home

## This is what happened instead:
Seemingly nothing then the error:
'this joint is already homed'

## It worked properly before this:
2.7 for sure. I can;t say when it broke yet

## Information about my hardware and software:

Fairly recent 2.8, AXIS gui
I don;t think it's linuxcnc it's self, as I can home using the python interface to linuxcnc (I have a panel that uses panelui programconnected at the sme time as AXIS.

andypugh, 2019-05-22:
No movement from the Z axis?

c-morley, 2019-05-22:
Ok this seems to be a configuration problem. Rod W has noticed that it doesn look right:

[TRAJ]

AXES = 3
COORDINATES = X Z
LINEAR_UNITS = mm
ANGULAR_UNITS = degree
DEFAULT_LINEAR_VELOCITY = 2.50
MAX_LINEAR_VELOCITY = 25.00

[KINS]
KINEMATICS = trivkins
#This is a best-guess at the number of joints, it should be checked
JOINTS = 3

this looks like a hold over from 2.7 when axes required 3 axes for a lathe rather then 2 axes now for a lathe?

c-morley, 2019-05-22:
I can confirm in my case it's a miss-configuration.
Would it be prudent to add an error if there is missing {JOINT] entrry ?

andypugh, 2019-06-19:
I have pushed a fix to the misconfiguration, 45421ebc9f8b131fcbfdf5bbf9f5e569ec5f90ba but haven't actually checked that it homes correctly. 
I will do that tonight.

andypugh, 2019-06-19:
I found one problem (axis.2.motor.... was being changed to joint.2.motor..... but that is now fixed, and this configuration (after conversion to a different Mesa card) does now home,

closed 2019-06-19 by andypugh