LinuxCNC/linuxcnc #598: Hostmot2 BISS MPRegs number check is wrong in abs_encoder.c

pcw-mesa, 2019-05-24:
MPregs for BISS is 7 but driver checks for 3

    case HM2_GTAG_BISS:
        if ( ! hm2_md_is_consistent_or_complain(hm2, md_index, 0, 3, 4, 0x03)) {
            HM2_ERR("inconsistent absenc Module Descriptor!\n");
            return -EINVAL;
        }

Should be
    case HM2_GTAG_BISS:
        if ( ! hm2_md_is_consistent_or_complain(hm2, md_index, 0, 3, 4, 0x07)) {
            HM2_ERR("inconsistent absenc Module Descriptor!\n");
            return -EINVAL;
        }

andypugh, 2019-05-24:
It's an easy fix. Do you think it needs doing it 2.7 or should I limit it to 2.8?

pcw-mesa, 2019-05-24:
Probably both, I think what happened is that I fixed the MPregs number in the firmware
a long time ago(years), but only non LinuxCNC users used the BISS interface so the mis-match 
was not noticed

pcw-mesa, 2019-05-24:
Ack its not as simple as I thought, the BISS  interface registers you designed around were modified
somewhere around 2014 to fix some non-linuxCNC customer issues so the driver needs changes
(the control/setup register was split into 2 registers to make space for a digital filter time setting field)

andypugh, 2019-05-24:
Can you tell me what to do, in simple words :-)

pcw-mesa, 2019-05-25:
Here's the current firmware register map: 
http://freeby.mesanet.com/biss
Looks (from the driver source) that the original mapping was:

data
control/setup reg
global start/status

New mapping is

data
controi/setup reg 0
control/setup reg 1
global start/status

I think the control/setup register data bits and are basically in the same places 
but spread to 2 registers to make room for the newly added digital filter register

I found a reasonably priced BISS encoder on Ebay i will purchase to test
the interface.

pcw-mesa, 2019-11-16:
I've have this fixed now (though only tested on one BISS encoder)
Just needs to be merged

andypugh, 2019-11-16:
I somehow lost this one. I did start working on it. Maybe I lost it on a VM somewhere. 
Is there a pull request?

pcw-mesa, 2019-11-18:
Not yet, I still see an issue with the precision of the position, I thought (perhaps wrongly) that the position was represented by double precision floating but It seems to be single precision. My test BISS encoder is 19 bits per turn and 11 bits of turn counts, so just moving 10 turns from 0 shows loss of precision (as if there are only ~ 6 digits of precision)

pcw-mesa, 2019-12-18:
OK, sorry  false alarm, the precision of the BISS encoder is fine, I was mislead by "watch" and halmeter that format the display number to fit available display space. The python hal interface
prinst the full number (thanks Jepler)

closed 2020-01-15 by pcw-mesa