LinuxCNC/linuxcnc #607: Jog Override

grandixximo, 2019-06-11:
## Here are the steps I follow to reproduce the issue:

 1. open LinuxCNC axis interface
 2. Jog + and Jog - from interface
 3. Change feed rate to 50%

## This is what I expected to happen:

The feed rate is slowed down to 50%

## This is what happened instead:

Feed rate is kept the same

## It worked properly before this:

after upgrading from 2.6

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`): latest master from github
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [x] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`): 2.9 pre
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc): Axis
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): sim

###############

Why can't jogging be override like in previous old release?

Would it be possible to have jog override pins? if not by interface it could be dealt with in HAL.

andypugh, 2019-06-11:
Having gone back in time and compiled a few versions of LinuxCNC I can confirm that this change happened between 2.6.13 and 2.7.0 around September 2015. 

The Axis UI has a jog-speed slider, what is it that you wish to achieve that the jog speed slider does not cover? 

Curiously a feed-override of zero _does_ prevent jogging.

grandixximo, 2019-06-12:
Been able to change feed while jogging from keyboard only, the jog speed slider does not effect speed while moving, basically on linux 2.6 you could use the keyboard like a pendant, you used the arrows to move and the numbers to adjust the feed, if i adjust the feed slider with the comma and point from the keyboard the speed doesn't change, have stop the move and start again, also would be useful to have jog feed override for other pendant implementation, there are work around for that, but a native jog feed override would be a nice touch IMO

c-morley, 2019-06-12:
I also agree feed override should work with jog rate.
Do we know why it changed?

andypugh, 2019-06-12:
c-morley: I was hoping you could tell us why it changed :-)
I have a feeling it was related to the decision to make feed-override not affect rapid speed:
https://sourceforge.net/p/emc/mailman/message/31438983/

It has been this way for the whole of the 2.7 version (and nobody complained until now...), so it might be a feature of the new trajectory planner.

andypugh, 2019-06-12:
Dewey found the discussions in the archive:

15sep2013 https://www.mail-archive.com/emc-developers@lists.sourceforge.net/msg11020.html
13nov2014 https://www.mail-archive.com/emc-developers@lists.sourceforge.net/msg14005.html

I find the second one interesting as it seems there was a concensus that jog speed should not be affected by feed override.  And I even describe crashing a tool because it did...

c-morley, 2019-06-12:
So yes we changed rapids to have it's own override separate from feedoverride.
I don't remember the discussion for making override not affect jogging.
I think one of the overrides should control jogging - it does seem useful.
I guess we could add 'jog override' and let the integrator decide...

Or maybe I need time to get used to the idea (I just got my lathe working again)

grandixximo, 2019-06-12:
In the old discussion i read some resistance to the change and a request for a ini file option, since there are different opinions on the utility of the feature, an option would be acceptable, I don't agree with the reasoning behind the INI option being not a viable solution, it would be awesome for me