LinuxCNC/linuxcnc #610: Freeze when executing endless loop of G4, no E-stop possible

aShure, 2019-06-13:
## Here are the steps I follow to reproduce the issue:
 1. execute g-code that contains a loop like
o100 while [ 1 eq 1]
G4 P100
o100 endwhile

## Result
Immediate freeze. Can't even E-stop.

andypugh, 2019-06-13:
I think that this is a (known) problem of the Axis graphical preview. 
Pressing "Esc" will eventually break out of the preview. 
The G-code will actually execute correctly. 
(and do nothing, for ever)

Only the GUI is frozen (or, technically, busy). HAL pins such as halui.estop.activate and motion.enable still have the correct effect so apart from the axis GUI LinuxCNC is fully functional underneath. 

Perhaps, though, a 10 second timeout in the preview would be sensible? 

If you _know_ that you have coded an infinite loop you can bypass this issue:

o100 while [ 1 eq 1]
G4 P100
(AXIS, stop)
o100 endwhile

I will leave this issue open for a while to see if there is any discussion on the idea of a timeout in the preview code.

aShure, 2019-06-14:
Great hint with the (AXIS, stop)
I accidentally crafted an infinite G4 loop in a subroutine I was working on for an ATC. Draw bar halfway up, tool stuck in pocket.. yes I would have welcomed either a timeout or axis not freezing ;)

The endeavor reminded me to use the hardware e-stop more often to not rely on the habit of using the software e-stop.

aShure, 2019-06-14:
same as #561

andypugh, 2019-06-14:
On Fri, 14 Jun 2019 at 16:37, Shure <notifications@github.com> wrote:

> Great hint with the (AXIS, stop)
> I accidentally crafted an infinite G4 loop in a subroutine I was working
> on for an ATC. Draw bar halfway up, tool stuck in pocket.. yes I would have
> welcomed either a timeout or axis not freezing ;)
>

Did you try "Esc". It might take a few seconds, but has always worked for
me.

-- 
atp

aShure, 2019-06-14:
Yes, I hit ESC. A lot. I think ESC will catch at the earliest after one G4 is done. I did not wait the full 100s  to check.(G4P#var, where #var was 100)

andypugh, 2019-06-15:
The freeze is in the graphical preview, so it doesn't actually execute the G4 P100.