LinuxCNC/linuxcnc #615: Increase default HAL_NAME_LEN currently 47 characters

manymachines, 2019-06-24:
This is an enhancement request, not a bug.

I'm programmatically building the signal names and using Python hal.connect().  I ran into the name is too long error e.g.: 
HAL: ERROR: signal name 'scopeinterface.microscope.rapid-scope-zero-link2' is too long

Here I'm building the name composed of two different component names and a pin name and it ended up greater than the max limit.

The limit appears to be defined in hal.h:  HAL_NAME_LEN

The request is to increase the default to say 128 if possible.

andypugh, 2019-06-25:
This would be possible, but perhaps there is some advantage on forcing a shorter limit on the signal names. 
They are meant to be human-readable. Is there perhaps an argument that very long names will be harder to distinguish? 

Additionally, there is a risk of 128 character HAL signal names line-wrapping in the text editor, not fitting in the halshow window, etc etc.

rene-dev, 2019-11-30:
I would also increase it to 128. especially on larger hal file might make things easier to read. requires some more changes, and increase in hal version number.
last update was few years ago: https://github.com/LinuxCNC/linuxcnc/commit/5735855f5e3c7456299f4e16931d11d63979ecfe

andypugh, 2020-04-08:
I suppose this would be one way to quiet the snprintf warnings in gcc-8.
Though I would prefer to do it the right way, ie do something with the snprintf return value and warn the user of pin-name (etc) truncation.