LinuxCNC/linuxcnc #621: hal.get_value() should not require the creation of a component

manymachines, 2019-07-03:
## Here are the steps I follow to reproduce the issue:

 1.  In a standalone Python script call hal.get_value(<pin>) without creating a component
 
I am running a standalone Python script for probing - invoked manually from the terminal.  My understanding of hal.get_value() is that it can be used to get the value of any pin, param or signal not only those coming from a userspace component. In fact it does work if I first create a dummy component.  The request is that hal.get_value() should not require the creation of a component.

h = hal.component("dummy")
h.ready()
...
    if hal.get_value('motion.probe-input'):
        print('Probe is tripped - is the probe plugged in?')
        return -1

## This is what I expected to happen:

I expected hal.get_value() to return the value of the pin.

## This is what happened instead:

An error message was returned:

RuntimeError: Cannot call before creating component

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`):
    Linux Mint 19.1 Tessa
 * I am using this kernel version (shown by `uname -a`):
    Linux platform 4.19.1-rt3 #1 SMP PREEMPT RT Sat Mar 2 19:53:32 EST 2019 
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [x] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):
    Branch 2.8, last built I think June 25
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):
    AXIS
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card):
    Mesa 5i25

c-morley, 2019-07-03:
To access HAL there needs to be a shared memory pointer. That is created when the component is made. That is why a component is required. I don't think there is any way around that. The only thing that could be done is hide the details of creating a dummy component and that doesn't seem like a good solution to me.

manymachines, 2019-07-03:
Thanks for confirming the component is required.  I will continue to create a dummy component in order to use hal.get_value() in my script.

c-morley, 2019-07-03:
I don't think this applies to you but anyone else reading:
If you have a reference to any other component, for instance if you are extending gladevcp with a handler files, you could use that component rather then build a dummy one.

closed 2019-07-03 by c-morley