LinuxCNC/linuxcnc #627: Shutdown Error 2.8 Debian 10

jethornton, 2019-07-25:
Fresh install of 2.8 on Debian 10 and every time I shut down the Axis simulator I get the LinuxCNC Errors screen.

Print file information:
RUN_IN_PLACE=no
LINUXCNC_DIR=
LINUXCNC_BIN_DIR=/usr/bin
LINUXCNC_TCL_DIR=/usr/lib/tcltk/linuxcnc
LINUXCNC_SCRIPT_DIR=
LINUXCNC_RTLIB_DIR=/usr/lib/linuxcnc/modules
LINUXCNC_CONFIG_DIR=
LINUXCNC_LANG_DIR=/usr/lib/tcltk/linuxcnc/msgs
INIVAR=inivar
HALCMD=halcmd
LINUXCNC_EMCSH=/usr/bin/wish8.6
LINUXCNC - 2.8.0~pre1
Machine configuration directory is '/home/john/linuxcnc/configs/sim.axis'
Machine configuration file is 'axis.ini'
INIFILE=/home/john/linuxcnc/configs/sim.axis/axis.ini
VERSION=1.1
PARAMETER_FILE=sim.var
TASK=milltask
HALUI=halui
DISPLAY=axis
COORDINATES=X Y Z
KINEMATICS=trivkins
Starting LinuxCNC...
Starting LinuxCNC server program: linuxcncsvr
Loading Real Time OS, RTAPI, and HAL_LIB modules
Starting LinuxCNC IO program: io
Starting HAL User Interface program: halui
Found file(lib): /usr/share/linuxcnc/hallib/core_sim.hal
Found file(lib): /usr/share/linuxcnc/hallib/sim_spindle_encoder.hal
Found file(lib): /usr/share/linuxcnc/hallib/axis_manualtoolchange.hal
Found file(lib): /usr/share/linuxcnc/hallib/simulated_home.hal
Found file(lib): /usr/share/linuxcnc/hallib/check_xyz_constraints.hal
Starting TASK program: milltask
Starting DISPLAY program: axis
Shutting down and cleaning up LinuxCNC...
task: 4973 cycles, min=0.000007, max=0.003897, avg=0.001060, 0 latency excursions (> 10x expected cycle time of 0.001000s)
Removing HAL_LIB, RTAPI, and Real Time OS modules
Removing NML shared memory segments

Debug file information:
Note: Using POSIX realtime
note: MAXV     max: 5.000 units/sec 300.000 units/min
note: LJOG     max: 5.000 units/sec 300.000 units/min
note: LJOG default: 0.250 units/sec 15.000 units/min
note: jog_order='XYZ'
note: jog_invert=set([])
python2: emc/rs274ngc/interp_setup.cc:189: setup::~setup(): Assertion `!pythis || Py_IsInitialized()' failed.
/usr/bin/linuxcnc: line 896:  1793 Aborted                 $EMCDISPLAY -ini "$INIFILE" $EMCDISPLAYARGS $EXTRA_ARGS
1742
1789
Stopping realtime threads
Unloading hal components
Note: Using POSIX realtime

Kernel message information:
/usr/lib/tcltk/linuxcnc/show_errors.tcl:
   dmesg: read kernel buffer failed: Operation not permitted

dmesg requires root privilege on this system
Description:	Debian GNU/Linux 10 (buster)
Use a terminal to issue the command: sudo dmesg

robEllenberg, 2019-07-25:
I think this is due to my recently-merged test branch. Among other things, the gcodemodule interpreter is now managed by a unique_ptr, so it will destroyed on shutdown. Before this, I don't think that interpreter's destructor was ever called (pinterp is never deleted in gcodemodule except during init). 

Does anyone know why Py_IsInitialized() is asserted in the setup_struct destructor? If it has to be there, then a simple fix is just to revert [7b28d7d3b40](https://github.com/LinuxCNC/linuxcnc/commit/7b28d7d3b40a1929e0b913153647588115a51fa9).

phillc54, 2019-07-30:
There have been reports of this on the forum as well. I see the same on Debian 9.

c-morley, 2019-08-05:
Did this get fixed? If it did can you please close the issue.

closed 2019-08-07 by robEllenberg