LinuxCNC/linuxcnc #629: hal status shows wrong spindle speed in constant surface speed

gmoccapy, 2019-08-07:
LinuxCNC status does not report the correct spindle speed, when in constant surface speed.

Just use a ngc file on any metric lathe config with 

G8 G18 G21 G40 G49 G54 G64 G90 M9 M5 G64P0.005 
G0 X100
G96 D1500 S100
M3
G95
G1 X10 F0.5
M2

Watch the spindle feedback, you will see it does increase as expected, watch also status spindle speed, it will show 1500 rpm, what is wrong! That does also afect the python modul, as it will not be updated properly.

It is in 2.7 ; 2.8 and master

Norbert

c-morley, 2019-08-08:
I believe that linuxcnc usually reports the requested speed, not the actual speed.
In constant surface feed you are saying it reports the maximum speed allowable?

Oh in 2.8 + Gstat will report the actual spindle speed as long as you connect something to HAL pin spindle.0.speed-in

rene-dev, 2019-08-10:
in css mode the requested speed changes, even with no encoder. so this is clearly a bug. the requested speed is send to the drive, so that value should be reported everywhere.

andypugh, 2019-08-10:
Might be difficult. I think the NML message is the css number, and the calculation is done as the HAL pin is set. There may be no existing path for that number back to the user space layers.

gmoccapy, 2019-08-12:
> Oh in 2.8 + Gstat will report the actual spindle speed as long as you connect something to HAL pin spindle.0.speed-in

I just looked at the code of GSTAT in 2.8 and I do not see any report of spindle speed. We do not need the spindle feedback, as the value we need is the commanded speed and that do change according to the dieter of the work piece.

c-morley, 2019-08-12:
actual-spindle-speed-changed is the message to watch for. I assume you are trying to do surfacefeed eater calculations or something similar. This would allow you to calculate the actual rate ad long as there is feedback connected. We could also change requested-spindle-speed message to watch the proper Hal to get what you want..as a work around.

gmoccapy, 2019-08-12:
Chris, it is not to find a work around, i have such a solution reading the spindle feedback bar of gmoccapy. The fact is, that linuxcnc should report the actual commanded spindle speed, not the maximum allowed one.
Imho the actual behavior is a bug!

c-morley, 2019-08-12:
Oh I agree. It just might be hard to fix. Using Gstat would not require anyone to connect a HAL pin..if we changed the requested-speed function. But yes ideally it should be reported correctly

andypugh, 2019-08-13:
Speed is calculated on-the-fly here, and put straight on the HAL pin. 
https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/motion/control.c#L1878
An element would need to be added to emcmotStatus to transfer that data back upstream to the userspace layers.