LinuxCNC/linuxcnc #632: V 2.9 ignoring joint.2.home-sw-in while homing

rodw-au, 2019-08-10:
## Here are the steps I follow to reproduce the issue:
 1. Previously running Master branch  v2.9
 2. Compiled latest version of 2.9 as of today
 3. Homed machine

## This is what I expected to happen:
YYXZ gantry Machine would home normally just as it did earlier today under earlier version of master branch V 2.9

## This is what happened instead:
Joint 2 home switch input was ignored and limit switch was triggered. This behavior persisted on several retries.

## It worked properly before this:
Yes was proven working on an earlier version of V2.9 immediatley prior the upgrade.

## Information about my hardware and software:
This is a long standing master branch (v2.8 and V2.9) gantry machine running Linux Mint 17.3 RT_PREMPT and Mesa 7i76e
Checked all signals were working for this homing  switch from the 7i76e input, to the signal and onto the joint pin.

andypugh, 2019-08-10:
Is this a buildbot install or do you compile it yourself? 
Homing was touched here: https://github.com/LinuxCNC/linuxcnc/commit/5ea573adde87231cdc566e3759f85bb4037869f7
I guess you don't know the version of 2.9 that was running before? (I wouldn't if anyone asked me)

If you are using a buildbot package then you can use "package -> force version to go back a step or two, and see at which point the problem goes away. The versions match the hashes here: https://github.com/LinuxCNC/linuxcnc/commits/master (so  "**5ea573a**" is the same as "1:2.9.0-pre0.433.**g5ea573a**dd (stable)". If building with git then you can check-out the specifi commit(or the one immediately before it) and build.

rodw-au, 2019-08-10:
Andy, thanks for your interest. As you say, I don't know what I was running prior to today's update as I just did what I thought  was a routine upgrade. As I said in step 2, I compiled from source (as always). If you can advise the correct commands to checkout the last commit prior to the suspected change, I am happy to  give it  go. Usually when stuff like this happens its a busted sensor or something similar. It not long since I replaced that homing sensor so I was pretty methodical confirming the  hardware was OK before raising the issue. I thought my next step was to roll back to V2.8

andypugh, 2019-08-10:
Rod. Not at a computer today. 
But google for “git bisect” instructions.

rodw-au, 2019-08-10:
Andy, thanks I'm a it confused now. When I look at commits, I see that commit 5ea573a shows as being committed yesterday but it was committed 22 days ago. So does that mean it was changed yesterday? I won't be near this machine until Monday.

andypugh, 2019-08-10:
I think that 5ea573a was probably committed to 2.8 (or 2.7) first and has only just been merged to 2.9.

rodw-au, 2019-08-13:
This can be closed. Now working. I don't know if it was a later commit or some config error in my repository.

closed 2019-08-13 by rodw-au