LinuxCNC/linuxcnc #638: Dev maillist (re-)subscribe without google AI

BsAtHome, 2019-09-02:
Hi,

I wanted to (re-)subscribe to the dev mailing list to discuss some issues with the RPI 3 and 4 hostmot SPI driver and build-environment, but the sourceforge lists require recaptcha junk. I block all google by default and am rather unwilling to do work for google's AI without getting handsomely paid by google (read: a rather wealthy amount of handsome until death do us part).

Is there an alternative route to re-subscribe or should I consider google's and souceforge's involvement impenetrable?

--
Greetings Bertho

andypugh, 2019-09-02:
On Mon, 2 Sep 2019 at 15:35, BsAtHome <notifications@github.com> wrote:

> Is there an alternative route to re-subscribe or should I consider
> google's and souceforge's involvement impenetrable?
>

Try sending an email to emc-developers@lists.sourceforge.com where both the
subject and the message body are just the word "subscribe"

I don't know for sure that that works, but it was the way to subscribe to
mailman lists last century.

The Captcha is Sourceforge, beyond LinuxCNC's control.
I have my own objections, as I don't know what a "Crosswalk" is, my Taxis
are black, school buses are red, and we don't have fire hydrants.
ie, the questions are really very US-centric.

-- 
atp

BsAtHome, 2019-09-02:
Yes, the old subscribe trick.
One problem though, the mailserver address is invalid. Have you got the "real" one?

"crosswalk" == anywhere; school bus: object not found; etc ;-)

<pre>   ----- The following addresses had permanent fatal errors -----
<emc-developers@lists.sourceforge.com>
    (reason: 550 Host unknown)

   ----- Transcript of session follows -----
550 5.1.2 <emc-developers@lists.sourceforge.com>... Host unknown (Name server: lists.sourceforge.com: host not found)
</pre>

andypugh, 2019-09-02:
On Mon, 2 Sep 2019 at 18:20, BsAtHome <notifications@github.com> wrote:

> Yes, the old subscribe trick.
> One problem though, the mailserver address is invalid. Have you got the
> "real" one?
>

My mistake, sourceforge.net, not .com
-- 
atp

BsAtHome, 2019-09-02:
Message was rejected (no post allowed). But, it referenced the emc-developers-owner@lists.sourceforge.net address, which I sent a message to, to do a manual subscribe if possible. Lets see if that works.

I guess, someone at the lcnc group is list admin, I hope.

BsAtHome, 2019-09-06:
No reply... I'll create an issue instead.

closed 2019-09-06 by BsAtHome

andypugh, 2019-09-06:
Sourceforge appear to use mailman. The mailman docs suggest some special mailing addresses:
http://www.list.org/mailman-member/node10.html

BsAtHome, 2019-09-06:
The "emc-developers-subscribe@lists.sourceforge.net" results in a "user unknown" error.
Using "emc-developers-join@lists.sourceforge.net" results in a reply pointing to the web-interface and we're back to the captcha problem.

Well, I created https://github.com/LinuxCNC/linuxcnc/issues/643 to ask about the problem.

andypugh, 2019-09-06:
I just got a reply to a "help" message  from emc-developers-request 

So try emc-developers-join@lists.sourceforge.net