LinuxCNC/linuxcnc #639: G33.1 rapid move not synchronized

papagno-source, 2019-09-03:
## Here are the steps I follow to reproduce the issue:

 1.run in MDI
 2.run m3s10
 3.run g33.1 z-50 k2
 4.when z axis move in negative direction press abort
 5.run m3s10
 6.run g33.1 z-50 k2

The axis z not wait synchronized with encoder , but go in radip move,  for stop in indeterminate position and continued after with reduce velocity for synchronized at encoder for finisch rigid tapping

The problem i have test in master2.8.pre0 and actual master 2.9.pre0

Can replace problem in axis , but have test on real machine is the same.

Very dangerous.

robEllenberg, 2019-09-05:
I see this too in the stock axis config. Interestingly, doing another motion after the abort (e.g. a rapid back to Z0) makes it behave normally. I suspect a static variable in the rigid tap routine in the TP is not being cleared correctly on abort.

papagno-source, 2019-09-06:
Yes confirm , if after abort send another motion command type g0 z0 , the g33.1 move correct.
If after abort , you move in manual every axis , the g33.1 have again problem, probably not is reset some variable.Only after motion command this variable is reseted.

andypugh, 2020-04-24:
Closed by b3fa2975fde671f10087cb85671e097f35c45d6b

closed 2020-04-24 by andypugh