LinuxCNC/linuxcnc #644: Joint 1 won't jog in joint mode on Axis 2.7.14 on linuxcnc-stretch-uspace-amd64

jrf2112swbellnet, 2019-09-07:
Running linuxcnc-stretch-uspace-amd64.iso. Interface card is Mesa 7i80HD.

Using generskins with 6-axis puma-style MotoMan arm. All parameters for link and joint definitions seem correct. I have created CAD model of the Motoman arm and the Vismach display tracks correctly and all 6 axes work perfectly in MDI mode. Issue only occurs in manual joint mode.

Expected: In Joint mode, Jog [+] and [-] buttons in Axis interface should cause each chosen joint to move and produce axisui.jog.[joint] true signals in Hal Meter. 

Actual: Axes 0, and 2-5 operate as expected. When the "1" joint radio button is selected, there is no resultant motion or Hal Meter output when jog buttons are clicked.

This did not occur on my previous setup (Wheezy). Might be related to this?
https://github.com/LinuxCNC/linuxcnc/issues/146

andypugh, 2019-09-08:
Can we see your HAL file. to eliminate a HAL error as the cause?

jrf2112swbellnet, 2019-09-09:
Certainly. I should have attached this in the first message. Apologies. I'm attaching a zip file with all the pertinent information that I'm aware of.

There isn't any hardware hooked up yet except for the Mesa card. It's using the 7i80hd_25_svst4_8.bit firmware, I believe.

[MotoManFiles.zip](https://github.com/LinuxCNC/linuxcnc/files/3590041/MotoManFiles.zip)

andypugh, 2019-09-09:
As this seems to be entirely a sim config it should be fairly easy for others to look at, at least.

jrf2112swbellnet, 2019-09-09:
In an attempt to troubleshoot, I modified a copy of the axis python script and forced the axis to 1 when jogging. That made axis 1 jog, but it also made all selected axes jog 1 too. I also noticed that after running the flowsnake g-code (with offset g56 to not crash the sim), it occasionally jogged axis 1 correctly. There didn't seem to be any consistency.

Thank you for looking at it. I spent many hours trying to get it to work. I have a FANUC sim that behaves the same way. That one is less complete.

Jim

andypugh, 2019-09-10:
I can confirm the problem. 
It is fixed in 2.8 as a side-effect of the Joints-Axes changes. 
I have had a look at how to fix it in 2.7, but I don't understand Tcl well enough.

jrf2112swbellnet, 2019-09-10:
Thank you for confirming. I was going crazy chasing a what I thought was a HAL issue.

I can switch to 2.8 but the path to do so is not as clear as 2.7. The stretch ISO made it easy. And I need the ability to drive a Mesa Ethernet card.

andypugh, 2019-09-10:
Dewey has a patch for this, but I haven't been able to try it yet. (And if it does work, I don't know how long it would take for it to be released as a 2.7 bugfix) 
As Python is interpreted you can try the change yourself without needing to recompile. 
It might even be possible to fix this using the .axisrc file, so maybe try that first. 
http://linuxcnc.org/docs/2.7/html/gui/axis.html#_axisrc
Create a file in your home diectory called ".axisrc" containing this line of Python code
`widgets.joint_1.configure(value="y")`
I don't guarantee that this will work, but if it does it's the easiest fix for you.

jrf2112swbellnet, 2019-09-10:
I will try that this evening. Thank you!

andypugh, 2019-09-10:
There is some indication that the same problem might be expected with axis C / 5

andypugh, 2019-09-10:
I tested it, and using a .axisrc works. 

With a text editor, create a new file, paste this in, and save is as .axisrc in your home folder. 
That should be a quick and easy fix. 

```
widgets.axis_c.configure(value="c")
widgets.joint_1.configure(value="y")
widgets.joint_5.configure(value="c")
```

jrf2112swbellnet, 2019-09-11:
That worked! Thank you very much. I am constantly amazed at the open source community. Very cool.

I will consider the problem solved.

ki4swy, 2020-07-24:
not sure if related, ( or in another bug report ) .. Compiled 2.9.0 lastnight , and have a 4 joint setup ( X Y Y Z ) , noticed the in Axis, Pressing Y would move Y motor 1, Pressing Z would move Y motor 2, and had no ability to move Z .. , this was not the case in GMOCCAPY.. Which appears to support the new Joint configuration...

andypugh, 2020-07-24:
If X, Y and Z move a joint then that sounds like a bug. If you are in joint mode you have joints 0, 1, 2 and 3. 
With most gantries jogging of the paired joints should be disabled in joint mode.

ki4swy, 2020-07-24:
Axis doesn't seem to give me the option to switch between world and joint mode, something maybe a miss.  but regardless, Z should not be moving JOINT 2 which is clearly under my [AXIS_Y] .. function works fine in GMOCCAPPY

andypugh, 2020-07-24:
The title of this bug report says "Joint Mode". Are you in Joint Mode? 
If you are in Joint mode then I would argue that Axis should not make assumptions about joint to axis mapping, and should do _nothing_ if you press X, Y or Z.