LinuxCNC/linuxcnc #645: Outputs energize twice when on button pressed in Gmocappy.

bobbevins, 2019-09-08:
## Here are the steps I follow to reproduce the issue:

 1. Pressing on button in Gmocappy causes halui.motion.is-on to pulse before actually going high and staying high.

## This is what I expected to happen: 

halui.motion.is-on should not pulse before turning on and staying on until the button is pressed again.

## This is what happened instead: 

Pressing the on button in gmocappy causes motion.motion-enabled to pulse before turning on. motion.motion-enabled triggers halui.motion.is-on.

This is with running gmocappy. On the same machine with the same config Axis gui does not have this behavior.

## It worked properly before this:
I do not know when this has changed.

## Information about my hardware and software:

 * I am using this Linux distributionDebian 9.9 Stretch:
 * I am using this kernel version 4.9.0-9-rt-amd64 SMP PREEMPT RT Debian):
 * I am running ...
   * RIP version

 * I am using this user interface gmocappy when this happens. Axis does not have this behavior.

gmoccapy, 2019-09-09:
If I am right, you do use gmoccapy 3.0.8 as this is the latest one. I just tested here in SIM Config with gmoccapy 3.0.8 and I can only reproduce that by clicking quite fast on the ON button.
Could you give more details how I can see that better than in hal show??

Norbert

bobbevins, 2019-09-09:
Look at it in halscope. use halui,motion.is-on or motion.motion.is-on.
You should see a pulse before it turns on. 
You can also look in forum there is a post with video and halscope from two people. I am not the only one that has this. Also I just tried the sim and it was doing it.

[https://forum.linuxcnc.org/38-general-linuxcnc-questions/37317-gmocappy-fires-relay-twice?start=10](url)

gmoccapy, 2019-09-10:
Hallo @ALL,

i am able to reproduce the behavior. so now is just to find the reason and bug fix that one.
I apologize for the inconvenience.

Norbert

gmoccapy, 2019-09-10:
OK, The error is not the code from the toggle button, but from hal status. Sending the command to turn on the machine send two hal status commands, first is off, second is on.
That will need some time to be fixed, so please be patient.

Norbert

c-morley, 2019-09-14:
It seems to be interaction between the code in toggle button and the two Hal status callbacks for on and off state..

I added debug prints:

    def on_tbtn_on_toggled(self, widget, data=None):
        print 'toggle on',widget.get_active(),self.stat.task_state == linuxcnc.STATE_ESTOP
        if widget.get_active():
            if self.stat.task_state == linuxcnc.STATE_ESTOP:
                widget.set_active(False)
                return
                (snip the rest of code was still here)

Then short circuited the hal_glib callback:

    def on_hal_status_state_on(self, widget):
        return

Now when you toggle the button you will sometimes see the screen be enabled then immediately turn off and you will see the debug prints confirm this. (It doesn't stay on because the hal_glib call back doesn't run)

Now short circuit hal_status_state_off too:

    def on_hal_status_state_off(self, widget):
        return

Now it toggles off and on in the debug statements.

So I think you need to rethink the code for turning linuxcnc on and off as right now the button does it and the hal_status does it too.

Take a look see if that helps you.

c-morley, 2019-09-14:
one idea I used in qtvcp is if the button is out of sync with the state of linuxcnc (for what ever reason) you toggle the button programmatically but block any code after that.
It's a little more complicated then that but hopefully I'm clean enough that you you get the idea.

gmoccapy, 2019-09-14:
Have you noticed, that estop reset sends a machine off signal?
And machine on seems to emit also the estop reset signal!

Norbert

gmoccapy, 2020-12-31:
solved with one of the later updates. I tested with 2.8 release and gmoccapy 3.1.0 and the rror is gone.

Norbert

closed 2020-12-31 by gmoccapy