LinuxCNC/linuxcnc #650: Homing single axis after failed homing homes more than one axis

grandixximo, 2019-09-29:
## Here are the steps I follow to reproduce the issue:

 1.Open linuxcnc with axis_9axis configuration
 2.Home and stop immediatly after by pressing ESC while Z is still homing
 3.select Z axis and press home button on keyboard to home single axis

## This is what I expected to happen:

Only Z axis homes

## This is what happened instead:

Z axis is homed and after all other axis home too

## It worked properly before this:
i think is something with new master linuxcnc 2.9 i don't remember having this issue on 2.7 or 2.8

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`): latest build from git
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [x] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`): linuxcnc-uspace_2.9.0~pre0_amd64
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc): AXIS

grandixximo, 2019-09-29:
I'd like to propose a fix

in the homing.c file, near the end of the file in the case HOME_ABORT need to add 

`joint_in_sequence[joint_num]=0;`

I think this is the correct fix

dngarrett, 2019-09-29:
Verified -- thanks for the report, the method to replicate using a sim config, and the proposed fix.  Bug exists in the 2.8 branch too, will fix

grandixximo, 2019-09-29:
You are welcome :-)

commit 5295eeeb (2019-09-29): homing.c additional cleanup after abort close #650
closed 2019-09-29 by dngarrett