LinuxCNC/linuxcnc #654: gmoccapy dro resets to g54 after starting program

jaapveeden, 2019-10-22:
## Here are the steps I follow to reproduce the issue:

 1. set coordinate system g55, g56 etc
 2. load a program without G54 like my_axis.ngc
 3. press cycle start
 4. DRO shows G54

## This is what I expected to happen:
DRO showing g56

## This is what happened instead:
 DRO shows g54 with the coordinates of g56

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`):
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 9.5 (stretch)
Release:	9.5
Codename:	stretch

 * I am using this kernel version (shown by `uname -a`):
Linux linuxcnc 4.9.0-8-rt-686-pae #1 SMP PREEMPT RT Debian 4.9.110-3+deb9u6 (2018-10-08) i686 GNU/Linux

 * I am running ...
   * [x] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):
2.7.14
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):
gmoccapy
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): mesa 7i92 card with 7i77 daughter card.

andypugh, 2019-10-23:
I suspect what you are seeing is the reset to G54 that happens at the end of the programme when the interpreter sees the M2. 
If you start and end the program with % and don't have an M2 then you will see the expected coordinate system on the DRO. 
If the G-code program is long enough, or contains tool changes (or other queue-busters) then you will see the coordinate system even with an M2. 

The reason for this is that the DRO is showing the interpreter state, rather than the motion system state. Very often the G-code has been completely interpreted and sent to the motion system queue before the tool even touches the work.

jaapveeden, 2019-10-23:
i will watch it with longer programs.
it makes sense sort off.. haha

thanks !!

closed 2019-10-23 by jaapveeden