LinuxCNC/linuxcnc #655: Motors left enabled after inversekinematics fail

ftkalcevic, 2019-10-24:
Whenever there is an kinematicsInverse error, axis appears to stop, but joint.n.amp-enable is still TRUE. This is on master.

This appears to be a bug in emc/motion/control.c get_pos_cmds().  If an error occurs, emcmotDebug->enabling is set to 0, and SET_MOTION_ENABLE_FLAG(0) is called.

This call to SET_MOTION_ENABLE_FLAG shouldn't be made.  This is done automatically in set_operating_mode().  Setting the flags causes set_operating_mode() to not work correctly.

SET_MOTION_ENABLE_FLAG(0) is called 4 times in get_pos_cmds().  They should all be deleted.
(Sorry - my github fork is broken, otherwise I would have submitted a pull request)

andypugh, 2020-01-15:
I have had this sat in my inbox since October, and have been meaning to have a look at it. 
(This is just to say that this hasn't been entirely lost from consideration)

andypugh, 2020-03-23:
Addressed by 252e2899f9611f4f1cc79dd9135991e97c8de4d8

closed 2020-03-23 by andypugh