LinuxCNC/linuxcnc #666: hm2/hm2_7c81.0: hm2_read_pin_descriptors: invalid port width 19

simonkuehling, 2019-12-11:
## Here are the steps I follow to reproduce the issue:

 1. LinuxCNC 2.9.0~pre0 (compiled from source on Raspbian with a Rapsberry Pi 3B+)
 2. Using a Mesa 7C81 AnythingIO FPGA board over SPI on the raspi
 3. Configured HostMot2 through
```
[HOSTMOT2]
DRIVER=hm2_rpspi
BOARD=7c81
```

## This is what I expected to happen:
From the report about the HostMot configuration of `57 I/O Pins used` it looks like the 7C81 does  have 19 pins per IO port set up - although HostMot2 seems to expect 24, 17 or 32 according to https://github.com/LinuxCNC/linuxcnc/blob/0f91c553a238c3f5e8a52285044761c2dcfd7de5/src/hal/drivers/mesa-hostmot2/pins.c#L613
Is this a bug in LinuxCNC or does the Mesa card report false information here?

## This is what happened instead:
LinuxCNC ist throwing a number of repeated errors during startup (full log attached as file):
```
hm2/hm2_7c81.0: hm2_read_pin_descriptors: invalid port width 19
hm2/hm2_7c81.0: hm2_read_pin_descriptors: invalid port width 19
hm2/hm2_7c81.0: hm2_read_pin_descriptors: invalid port width 19
[...]
```
[linuxcnc_terminal_log.txt](https://github.com/LinuxCNC/linuxcnc/files/3949978/linuxcnc_terminal_log.txt)

## Information about my hardware and software:

 * I am using this Linux distribution and version: Raspbian GNU/Linux 10 (buster)
 * I am using this kernel version: Linux raspberrypi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [X] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version: v2.9.0-pre0-872-g2118e13fa
 * I am using this user interface (GUI): AXIS
 * I am using this interface hardware vendor and chipset: Mesa 7C81 FPGA board via SPI on Raspberry Pi 3B+

simonkuehling, 2019-12-11:
Seems like the Mesa 7C81 configurations all do indeed have 19 pins per port set - 17 I/O pins and two serial pins (from the two RJ-45 connectors for RS-422/RS-485 interface)

The virtual ports reported from the FPGA firmware are named `P1+Serial`, `P2+Serial` and `P7+Serial` accordingly.
mesaflash --readhmid lists:
```
$ mesaflash --device 7C81 --spi --addr /dev/spidev0.0 --readhmid
Configuration Name: HOSTMOT2

General configuration information:

  BoardName : MESA7C81
  FPGA Size: 9 KGates
  FPGA Pins: 144
  Number of IO Ports: 3
  Width of one I/O port: 19
  Clock Low frequency: 100.0000 MHz
  Clock High frequency: 200.0000 MHz
  IDROM Type: 3
  Instance Stride 0: 4
  Instance Stride 1: 64
  Register Stride 0: 256
  Register Stride 1: 256

[...]

Configuration pin-out:

IO Connections for P1+Serial
Pin#  I/O   Pri. func    Sec. func       Chan      Pin func        Pin Dir

 0      0   IOPort       PWM              0        PWM             (Out)
 0      1   IOPort       None           
 0      2   IOPort       StepGen          0        Step/Table1     (Out)
 0      3   IOPort       None           
 0      4   IOPort       StepGen          0        Dir/Table2      (Out)
 0      5   IOPort       None           
 0      6   IOPort       StepGen          1        Step/Table1     (Out)
 0      7   IOPort       None           
 0      8   IOPort       StepGen          1        Dir/Table2      (Out)
 0      9   IOPort       StepGen          2        Step/Table1     (Out)
 0     10   IOPort       StepGen          2        Dir/Table2      (Out)
 0     11   IOPort       StepGen          3        Step/Table1     (Out)
 0     12   IOPort       StepGen          3        Dir/Table2      (Out)
 0     13   IOPort       None           
 0     14   IOPort       QCount           0        Quad-A          (In)
 0     15   IOPort       QCount           0        Quad-B          (In)
 0     16   IOPort       QCount           0        Quad-IDX        (In)
 0     17   IOPort       SSerial          0        TXData0         (Out)
 0     18   IOPort       SSerial          0        TXData1         (Out)

IO Connections for P2+Serial
Pin#  I/O   Pri. func    Sec. func       Chan      Pin func        Pin Dir

 0     19   IOPort       PWM              1        PWM             (Out)
 0     20   IOPort       None           
 0     21   IOPort       StepGen          4        Step/Table1     (Out)
 0     22   IOPort       None           
 0     23   IOPort       StepGen          4        Dir/Table2      (Out)
 0     24   IOPort       None           
 0     25   IOPort       StepGen          5        Step/Table1     (Out)
 0     26   IOPort       None           
 0     27   IOPort       StepGen          5        Dir/Table2      (Out)
 0     28   IOPort       StepGen          6        Step/Table1     (Out)
 0     29   IOPort       StepGen          6        Dir/Table2      (Out)
 0     30   IOPort       None           
 0     31   IOPort       None           
 0     32   IOPort       None           
 0     33   IOPort       None           
 0     34   IOPort       None           
 0     35   IOPort       None           
 0     36   IOPort       SSerial          0        TXEn0           (Out)
 0     37   IOPort       SSerial          0        TXEn1           (Out)

IO Connections for P7+Serial
Pin#  I/O   Pri. func    Sec. func       Chan      Pin func        Pin Dir

 0     38   IOPort       PWM              2        PWM             (Out)
 0     39   IOPort       None           
 0     40   IOPort       StepGen          7        Step/Table1     (Out)
 0     41   IOPort       None           
 0     42   IOPort       StepGen          7        Dir/Table2      (Out)
 0     43   IOPort       None           
 0     44   IOPort       StepGen          8        Step/Table1     (Out)
 0     45   IOPort       None           
 0     46   IOPort       StepGen          8        Dir/Table2      (Out)
 0     47   IOPort       StepGen          9        Step/Table1     (Out)
 0     48   IOPort       StepGen          9        Dir/Table2      (Out)
 0     49   IOPort       None           
 0     50   IOPort       None           
 0     51   IOPort       None           
 0     52   IOPort       None           
 0     53   IOPort       None           
 0     54   IOPort       None           
 0     55   IOPort       SSerial          0        RXData0         (In)
 0     56   IOPort       SSerial          0        RXData1         (In)
```

So this needs to be accounted for in the hostmot2 code, I guess?

rene-dev, 2019-12-11:
can you try again?

simonkuehling, 2019-12-11:
@rene-dev wow, that was quick!
My current setup is running on the master branch - but I cherry-picked your commit from there, recompiled and the error is gone.
Thanks!

simonkuehling, 2019-12-11:
Further testing proved the board is running just fine now, feeling confident closing this issue 👍

commit 2a42439e (2019-12-11): fix #666 by allowing pin width of 19
closed 2019-12-11 by simonkuehling