LinuxCNC/linuxcnc #668: Acceleration axis not correct with esternal offset

papagno-source, 2019-12-30:
# The issue tracker is not a support forum
 This problem is present for every axis 
If have value Max_ACCELERATION = 300
and have  OFFSET_AV_RATIO = 0.5
the real acceleration is MAX_ACCELERATION/OFFSET_AV_RATIO = 150
This problem is present in auto mode execution.

I thing no correct this, the OFFSET_AV-RATIO, it should only intervene when I use the offset function and not in the normal work cycle

(delete this section before submitting your bug report)

## Here are the steps I follow to reproduce the issue:

 1.open linuxcnc
 2.start exsternal offset example ini file 
 3.impose OFFSET_AV_RATIO 0.5
4. the acceleration in automatic work will 1/2 the impose in MAX_ACCELERATION ini file.
5.repeat change value OFFSET_AV_RATIO 0.9
6.start in auto the same program and axis move slow accelleration .
7.repeat change value OFFSET_AV_RATIO 0.1
8.6.start in auto the same program and axis move with similar acceleration the value  MAX_ACCELERATION

## This is what I expected to happen:

## This is what happened instead:

## It worked properly before this:
## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`):
acutual master
 * I am using this kernel version (shown by `uname -a`):
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card):

dngarrett, 2019-12-30:
The [AXIS_L]OFFSET_AV_RATIO is read only once at startup.

> it should only intervene when I use the offset function and not in the normal work cycle

There is no "offset function", the startup value applies for all world (teleop) jogs and coordinated (gcode) motion (auto or mdi).

papagno-source, 2019-12-30:
Yes indeed.
It would no longer be fair to have a scaled acceleration of the factor [AXIS_L] OFFSET_AV_RATIO only when I have to use the external offset function and instead have the acceleration that I want and that I have set in the [AXIS_L] MAX_ACCELERATION parameter for the jobs where I want the real machine performance .
With a value of [AXIS_L] OFFSET_AV_RATIO medium / high, in order, for example, to reduce the vibrations on the axes, while I move in the handwheel, when I have enabled the external offset function, even the real processing will slow down, due to the fact that the accelerations of the axes are reduced, compared to those desired in [AXIS_L] MAX_ACCELERATION.
Could it be changed?

papagno-source, 2019-12-30:
261/5000
I would use [AXIS_L] OFFSET_AV_RATIO when I have to use the Jog Wheel, that is, manually to move with the jog wheel and when I use the external offset function, but in the car the acceleration should be that programmed in [AXIS_L] MAX_ACCELERATION.

dngarrett, 2019-12-30:
The ini values for *some* settings can be changed with 'inihal' pins.  Included are: ini.N.max_velocity, ini.N.max_acceleration for joints and ini.L.max_velocity, ini.L.max_acceleration for world (xyzabcuvw) coordinates.

See: http://linuxcnc.org/docs/2.8/html/man/man1/milltask.1.html

NOTE restrictions in the manpage-- 1) inihal changes are not dynamic and cannot affect ongoing gcode motion, 2) set with postgui halfiles or [APPLICATION]APPS but not with [HAL]HALFILE=.  One usually must provide virtual panels or concurrent applications to manage these hal pins.

For JOGGING, there are existing hal pins to limit acceleration when jogging joint.N.jog-accel-fraction and axis.L.jog-accel-fraction

Ref: http://linuxcnc.org/docs/2.8/html/man/man9/motion.9.html

papagno-source, 2019-12-30:
In fact, since we cannot change the acceleration value, via the .ini pin, in the running state, and using the external offset, in pause and auto mode, it would be advisable to perform this automatic change of the acceleration value from the source.
Where can I, in the code, try to understand how to impose the acceleration [AXIS_L] OFFSET_AV_RATIO only in exsternal offset enable and reset the [AXIS_L] MAX_ACCELERATION. when do I not use exsternal offset enable?

dngarrett, 2019-12-30:
I haven't really been able to understand your use-case but for many cases it is sufficient to just alter the inihal pins when the interpreter is not active -- changes will be honored in the next program.

Suggest you experiment with (X example):
$ sim_pin ini.X.max_acceleration ini.X.max_velocity &

> "since we cannot change the acceleration value, via the .ini pin, in the running state"

This is not strictly true.  The ini accel/vel values cannot be changed while planned, coordinated moves are *in_progress* but can be changed in between program startups.  Moreover, if a program includes synchronization events (queuebusters), inihal pin value changes will be honored on the moves planned after synchronization.  Syncing events include tool changes and hal input reads like M66.  This means one could insert multiple immediate reads (like M66POLO) as synchronization points in a program to allow recognition of inihal pins at numerous times.

Background:

External offsets are applied using simple_tp -- a single coordinate planner that employs a trapezoidal velocity profile and follows the per-axis constraints allocated by the axis accel/vel limits as ratioed by the OFFSET_AV_RATIO setting.  Coordinated motion (gcode) is planned by the multidimensionl trajectory planner constrained by the accel/vel limits of multiple coordinates as well as the overall traj limits.  So, when coordinated motion is occurring, the multidimensional trajectory planner has already used the *allocated* accel/vel limits that were valid at the *initiation* of the trajectory planning.  If one were to then allow an increase in the limits used by the external offsets single-dimension planner during such motion, violations of the machine limits could not be prevented.  

In some applications (plasma torch height control) coordinated motion and external offsetting of the controlled axis are *mutually exclusive* so users can set the accel/vel limits to twice the machine limit and use an offset ratio of 0.5.  

If you want to change the code for your appliction, most changes
occurred in a single big commit:

https://github.com/LinuxCNC/linuxcnc/commit/784db03355

papagno-source, 2019-12-31:
Ok . thanks for reply. i ceck the code, but not know more the C language.
Another futhure for example can is improve rapid override value for offset clear position, now axis return in rapid and override rapid not have effect.
Thanks again.

closed 2020-01-13 by dngarrett