LinuxCNC/linuxcnc #675: component thcud.comp

nkp216, 2020-01-21:
1.
pin cur_offset not connected anywhere

When I needed it, I did it like this:
................
................
    if(!enable){
        z_pos_out = z_pos_in;
        z_fb_out = z_pos_in; // keep axis motor position fb from being confused
    }
    cur_offset = offset;
}

2. 
And it would be nice to make parameter "correction_vel"
a pin so that you can adjust from the outside

like thise:
pin in float correction_vel "The Velocity to move Z to correct";

andypugh, 2020-01-21:
I see no reason that correction_vel could not be a pin. It won't affect any existing HAL files and sounds like it might be useful. 
I don't see any reason for the offset to be an internal variable. Your solution seems like it would have the desired effect, but it would make more sense to use the pin value directly throughout. 
Hopefully JT can explain if there is a good reason for the separation.

petterreinholdtsen, 2023-10-03:
@jethornton Any chance you can explain the reason for the separation?

jethornton, 2023-10-03:
It looks like:
pin out float cur_offset "The Current Offset";
is a left over from the copy and paste of thc.comp perhaps but I can't find any reference to it. So maybe I was adding it 10 years ago and never finished it...

JT

andypugh, 2023-10-03:
I have made the change to the offset output pin.

closed 2023-10-03 by andypugh