LinuxCNC/linuxcnc #682: Classicladder:4229: Gtk-CRITICAL **: IA__gtk_widget_get_parent:

Hortensie, 2020-02-13:
## Here are the steps I follow to reproduce the issue:

 1.Set debug level to 3, load classic ladder with modbus
 2. run linucnc from command line

## This is what I expected to happen:
No Crtical errors in debug logs

## This is what happened instead:
LINUXCNC - 2.9.0-pre0-774-ge67a63cb7
Machine configuration directory is '/home/devil/linuxcnc/configs/AXIS_X_MODBUS'
Machine configuration file is 'AXIS_X_MODBUS.ini'
Starting LinuxCNC...
Found file(REL): ./AXIS_X_MODBUS.hal
Note: Using POSIX realtime
hm2: loading Mesa HostMot2 driver version 0.15
hm2_pci: loading Mesa AnyIO HostMot2 driver version 0.7
hm2_pci: discovered 5i25 at 0000:08:00.0
hm2/hm2_5i25.0: Smart Serial Firmware Version 43
Board hm2_5i25.0.7i76.0.0 Hardware Mode 0 = standard
Board hm2_5i25.0.7i76.0.0 Software Mode 0 = io_spin
Board hm2_5i25.0.7i76.0.0 Software Mode 1 = io_ana_spin
Board hm2_5i25.0.7i76.0.0 Software Mode 2 = io_enc_ana_spin_fv
hm2/hm2_5i25.0: 34 I/O Pins used:
hm2/hm2_5i25.0:     IO Pin 000 (P3-01): StepGen #0, pin Direction (Output)
hm2/hm2_5i25.0:     IO Pin 001 (P3-14): StepGen #0, pin Step (Output)
hm2/hm2_5i25.0:     IO Pin 002 (P3-02): StepGen #1, pin Direction (Output)
hm2/hm2_5i25.0:     IO Pin 003 (P3-15): StepGen #1, pin Step (Output)
hm2/hm2_5i25.0:     IO Pin 004 (P3-03): StepGen #2, pin Direction (Output)
hm2/hm2_5i25.0:     IO Pin 005 (P3-16): StepGen #2, pin Step (Output)
hm2/hm2_5i25.0:     IO Pin 006 (P3-04): StepGen #3, pin Direction (Output)
hm2/hm2_5i25.0:     IO Pin 007 (P3-17): StepGen #3, pin Step (Output)
hm2/hm2_5i25.0:     IO Pin 008 (P3-05): StepGen #4, pin Direction (Output)
hm2/hm2_5i25.0:     IO Pin 009 (P3-06): StepGen #4, pin Step (Output)
hm2/hm2_5i25.0:     IO Pin 010 (P3-07): Smart Serial Interface #0, pin TxData0 (Output)
hm2/hm2_5i25.0:     IO Pin 011 (P3-08): Smart Serial Interface #0, pin RxData0 (Input)
hm2/hm2_5i25.0:     IO Pin 012 (P3-09): IOPort
hm2/hm2_5i25.0:     IO Pin 013 (P3-10): IOPort
hm2/hm2_5i25.0:     IO Pin 014 (P3-11): Encoder #0, pin Index (Input)
hm2/hm2_5i25.0:     IO Pin 015 (P3-12): Encoder #0, pin B (Input)
hm2/hm2_5i25.0:     IO Pin 016 (P3-13): Encoder #0, pin A (Input)
hm2/hm2_5i25.0:     IO Pin 017 (P2-01): Smart Serial Interface #1, pin RxData0 (Input)
hm2/hm2_5i25.0:     IO Pin 018 (P2-14): Smart Serial Interface #1, pin RxData1 (Input)
hm2/hm2_5i25.0:     IO Pin 019 (P2-02): Smart Serial Interface #1, pin RxData2 (Input)
hm2/hm2_5i25.0:     IO Pin 020 (P2-15): Smart Serial Interface #1, pin RxData3 (Input)
hm2/hm2_5i25.0:     IO Pin 021 (P2-03): Smart Serial Interface #1, pin TxData0 (Output)
hm2/hm2_5i25.0:     IO Pin 022 (P2-16): Smart Serial Interface #1, pin TxData1 (Output)
hm2/hm2_5i25.0:     IO Pin 023 (P2-04): Smart Serial Interface #1, pin TxData2 (Output)
hm2/hm2_5i25.0:     IO Pin 024 (P2-17): Smart Serial Interface #1, pin TxData3 (Output)
hm2/hm2_5i25.0:     IO Pin 025 (P2-05): Smart Serial Interface #1, pin RxData4 (Input)
hm2/hm2_5i25.0:     IO Pin 026 (P2-06): Smart Serial Interface #1, pin RxData5 (Input)
hm2/hm2_5i25.0:     IO Pin 027 (P2-07): Smart Serial Interface #1, pin RxData6 (Input)
hm2/hm2_5i25.0:     IO Pin 028 (P2-08): Smart Serial Interface #1, pin RxData7 (Input)
hm2/hm2_5i25.0:     IO Pin 029 (P2-09): Smart Serial Interface #1, pin TxData4 (Output)
hm2/hm2_5i25.0:     IO Pin 030 (P2-10): Smart Serial Interface #1, pin TxData5 (Output)
hm2/hm2_5i25.0:     IO Pin 031 (P2-11): Smart Serial Interface #1, pin TxData6 (Output)
hm2/hm2_5i25.0:     IO Pin 032 (P2-12): Smart Serial Interface #1, pin TxData7 (Output)
hm2/hm2_5i25.0:     IO Pin 033 (P2-13): Smart Serial Interface #1, pin TxEn7   (Output)
hm2/hm2_5i25.0: registered
hm2_5i25.0: initialized AnyIO board at 0000:08:00.0
creating ladder-state
Found file(REL): ./custom.hal
INFO CLASSICLADDER---I/O modbus master closed!
INFO CLASSICLADDER---I/O modbus master Data bits 8 Stop bits 1 Parity 0
INFO CLASSICLADDER---I/O modbus master (Serial) init ok !
emcTrajSetJoints(3) returned 0
emcTrajSetSpindles(1) returned 0
emcTrajSetAxes(3, 7)
emcTrajSetUnits(1.0000, 1.0000)
emcTrajSetVelocity(0.0000, 2.5000) returned 0
emcTrajSetMaxVelocity(25.0000) returned 0
emcTrajSetAcceleration(999999999999999967336168804116691273849533185806555472917961779471295845921727862608739868455469056.0000) returned 0
emcTrajSetMaxAcceleration(999999999999999967336168804116691273849533185806555472917961779471295845921727862608739868455469056.0000)
emcTrajSetHome(0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000) returned 0
emcJointSetType(0, 1)
emcJointSetUnits(0, 1.0000)
emcJointSetBacklash(0, 0.0000) returned 0
emcJointSetMinPositionLimit(0, -0.0100) returned 0
emcJointSetMaxPositionLimit(0, 200.0000) returned 0
emcJointSetFerror(0, 10.0000) returned 0
emcJointSetMinFerror(0, 1.0000) returned 0
emcJointSetHomingParams(0, 0.0000, 0.0000, -1.0000, 0.0000, 0.0000, 0, 0, 0, 999, 0) returned 0
emcJointSetMaxVelocity(0, 3.3333) returned 0
emcJointSetMaxAcceleration(0, 300.0000) returned 0
emcJointActivate(0) returned 0
emcJointSetType(1, 1)
emcJointSetUnits(1, 1.0000)
emcJointSetBacklash(1, 0.0000) returned 0
emcJointSetMinPositionLimit(1, -0.0100) returned 0
emcJointSetMaxPositionLimit(1, 200.0000) returned 0
emcJointSetFerror(1, 10.0000) returned 0
emcJointSetMinFerror(1, 1.0000) returned 0
emcJointSetHomingParams(1, 0.0000, 0.0000, -1.0000, 0.0000, 0.0000, 0, 0, 0, 999, 0) returned 0
emcJointSetMaxVelocity(1, 25.0000) returned 0
emcJointSetMaxAcceleration(1, 750.0000) returned 0
emcJointActivate(1) returned 0
emcJointSetType(2, 1)
emcJointSetUnits(2, 1.0000)
emcJointSetBacklash(2, 0.0000) returned 0
emcJointSetMinPositionLimit(2, -100.0000) returned 0
emcJointSetMaxPositionLimit(2, 0.0100) returned 0
emcJointSetFerror(2, 10.0000) returned 0
emcJointSetMinFerror(2, 1.0000) returned 0
emcJointSetHomingParams(2, 0.0000, 0.0000, -1.0000, 0.0000, 0.0000, 0, 0, 0, 999, 0) returned 0
emcJointSetMaxVelocity(2, 25.0000) returned 0
emcJointSetMaxAcceleration(2, 750.0000) returned 0
emcJointActivate(2) returned 0
emcAxisSetMinPositionLimit(0, -0.0100) returned 0
emcAxisSetMaxPositionLimit(0, 200.0000) returned 0
emcAxisSetMaxVelocity(0, 3.3333) returned 0
emcAxisSetMaxAcceleration(0, 300.0000) returned 0
emcAxisSetLockingJoint(0, -1) returned 0
emcAxisSetMinPositionLimit(1, -0.0100) returned 0
emcAxisSetMaxPositionLimit(1, 200.0000) returned 0
emcAxisSetMaxVelocity(1, 25.0000) returned 0
emcAxisSetMaxAcceleration(1, 750.0000) returned 0
emcAxisSetLockingJoint(1, -1) returned 0
emcAxisSetMinPositionLimit(2, -100.0000) returned 0
emcAxisSetMaxPositionLimit(2, 0.0100) returned 0
emcAxisSetMaxVelocity(2, 25.0000) returned 0
emcAxisSetMaxAcceleration(2, 750.0000) returned 0
emcAxisSetLockingJoint(2, -1) returned 0
INFO CLASSICLADDER-   Modbus I/O module to send: Lgt=8 <-  Slave address-2  Function code-6  Data-0 1 0 0 D8 39 
INFO CLASSICLADDER-   MODBUS Length we should receive=5 (+3)
Serial config...
Serial writing...
Writing done!
select() for serial reading...
**(classicladder:4229): Gtk-CRITICAL **: IA__gtk_widget_get_parent: assertion 'GTK_IS_WIDGET (widget)' failed**
note: MAXV     max: 25.000 units/sec 1500.000 units/min
note: LJOG     max: 25.000 units/sec 1500.000 units/min
note: LJOG default: 6.000 units/sec 360.000 units/min
note: jog_order='XYZ'
note: jog_invert=set([])
ERROR CLASSICLADDER-  MODBUS NO RESPONSE (Errs=1/1) !?
INFO CLASSICLADDER-   Modbus I/O module received: Lgt=0 -> 
ERROR CLASSICLADDER-   MODBUS-LOW LEVEL ERROR IN RESPONSE!!!
Serial flush all!

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`):
**Distributor ID:	Debian
Description:	Debian GNU/Linux 9.11 (stretch)
Release:	9.11
Codename:	stretch**

 * I am using this kernel version (shown by `uname -a`):
**Linux Hortes 4.9.0-11-rt-amd64 #1 SMP PREEMPT RT Debian 4.9.189-3+deb9u2 (2019-11-11) x86_64 GNU/Linux**

 * I am running ... 
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`): **LINUXCNC - 2.9.0-pre0-774-ge67a63cb7**
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc): **AXIS**
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card):  **FPGA Mesa 5i25 + 7i76**

c-morley, 2020-02-13:
Is that an actual problem with classicladder or is it just a scary sounding error message?

Hortensie, 2020-02-14:
At least in my case classic ladder fail to establish modbus communication. I was wondering if this is a root cause or not.

c-morley, 2020-02-14:
Pretty sure not as I can see the modbus error messages still complaining.

Hortensie, 2020-02-19:
Right, I resolved modbus issue. If you think that Critical warning in debug logs can be ignored we can close this issue.

c-morley, 2020-02-21:
Ok thanks for the update.

closed 2020-02-21 by c-morley