LinuxCNC/linuxcnc #690: Building master under Mint 19.2 (ubuntu 18.04)

ozzyrob, 2020-02-27:
I apologise for anything left out or lacking info, this my first bug report.
Rob

## Here are the steps I follow to reproduce the issue:

 1. clone linuxcnc
 2. check out master
 3. cd into src
 4. run ./autogen.sh
 5. run ./configure --with-realtime=uspace 
 6. run make

## This is what I expected to happen:
everything build with no errors

## This is what happened instead:
snippet of build error
```
emc/rs274ngc/interp_g7x.cc: In instantiation of ‘void swapped_motion<swap>::straight_move(std::complex<double>) [with int swap = 2]’:
emc/rs274ngc/interp_g7x.cc:1230:1:   required from here
emc/rs274ngc/interp_g7x.cc:578:41: error: ‘conj’ was not declared in this scope
  case 5: orig->straight_move(conj(1i*end)); break;
```

## It worked properly with this commit:
v2.9.0-pre0-1042-gdef92fbb4

## Information about my hardware and software:

 * I am using this Linux distribution and version 

```
No LSB modules are available.
Distributor ID:	LinuxMint
Description:	Linux Mint 19.2 Tina
Release:	19.2
Codename:	tina
```

:
 * I am using this kernel version (shown by `uname -a`):
`Linux VBox-M90P 4.16.12-rt5-kona-rt #1 SMP PREEMPT RT Fri Jun 8 10:39:47 JST 2018 x86_64 x86_64 x86_64 GNU/Linux`

 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):v2.9.0-pre0-1130-g86664852d

* I am using  g++ --version
```
g++ (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

```
```
g++ -print-libgcc-file-name 
/usr/lib/gcc/x86_64-linux-gnu/7/libgcc.a
```

andypugh, 2020-02-28:
Sorry about this, a commit last night that introduced G71 cycles for lathe roughing turned out to be problematic with some older compilers. 
There should be a fix in place before the weekend. 
Or try compiling the andypugh/g71_test2 branch if you want the new cycle.

ozzyrob, 2020-02-28:
That branch ,andypugh/g71_test2, built fine and was able to run axis sim for mill & lathe.
Someone asked on the forum about building linuxcnc on the Mint images I remastered and I came up with this issue when running through the procedure. 
Thanks for all your help. Just wish I was a little more proficient in programming and was able to help out with some of the hard yakka or hard graft.

closed 2020-05-02 by rene-dev