LinuxCNC/linuxcnc #691: Hazardous bug in jogging

pcw-mesa, 2020-02-28:
Pumuckl reported this in the LinuxCNC forum

If you are doing a continuous jog (+ or - button pressed) and type I for incremental jog, 
the motion continues even after you release the + or - button and continues to the soft limits

The OP was using Gmoccapy and I tested with Axis, same result 

Once in incremental mode the + and -  buttons are ignored,  though you can stop the 
runaway jog by changing back to continuous mode and pressing and releasing a + or - button

c-morley, 2020-02-28:
I tried this in master AXIS and can't convince my self if it jogs forever or just the length of the increment. if you press the +- button while it is moving it will add increments to the move making it seem to jog forever - I think.
probably adding a jog stop command every time the increment changes would cover this problem.

pcw-mesa, 2020-02-28:
It will jog to the limits  if you
1. Set jog mode to continuous (type C)
2. Press + or - button
3. While + or - button is pressed, type I

jethornton, 2020-02-29:
There is no problem with switching from incremental to continuous only 
when switching from continuous to incremental while jogging

c-morley, 2020-02-29:
In my  sim I can't get this to happen - I can occasionally get it to run on by an increment length.
Oh wait I think you meant use the GUI + button not the keyboard + button.
Yes now I can make it fail.

I'm pretty sure the problem still is based on how button/keyboard jogging works.
When jogging continuously the button 'release' stops jogging
When incremental jogging the button 'release' is ignored because the jog will stop when the increment is done.
So I bet that if you start a continuous jog then switch to increments then the jogging function ignores the button release so the jog continues.
So this is not a motion problem but a screen problem and surely all the screens that do jogging while allowing the increments to change could have the same behavior.

c-morley, 2020-03-02:
Fooling with this a bit in AXIS change the function:

def jogspeed_listbox_change(dummy, value):
    jjogmode = get_jog_mode()      #<--new line
    for jnum in range(num_joints):   #<--new line
        jog_after[jnum] = None           #<--new line
        jogging[jnum] = 0                    #<--new line
        jog(linuxcnc.JOG_STOP, jjogmode, jnum) #<--new line
    global jogincr_index_last

then recompile.
Now when you press the i key (while continuous jogging with gui button) the jog will stop.
But it's not quite enough - as when you release the gui button the axis will jog the now selected increment.

c-morley, 2020-04-04:
So it seems my code would have worked fine except that with button jogging in increments mode, the jogging starts with button release rather then button press. So my code interrupts the continuous jog when the increments setting changes but the incremental jog is initiated with the release of the button.
Keyboard jogging is the opposite and that is why it works fine.

But I can't seem to find how screen buttons call functions. Any one give me a clue?

c-morley, 2020-04-04:
I found it axis.tcl - why isn't that in the same folder as axis.py...

phillc54, 2020-04-04:
I think jogging should start on button press.

c-morley, 2020-04-04:
Yes. Looks like incremental jogging was handled differently in the past and this behavior is left over.
This seems to fix it:
https://github.com/LinuxCNC/linuxcnc/commits/axis_runaway_jog_fix
If someone else could confirm (you could just edit your existing code base it's very small) then I will push to 2.8

phillc54, 2020-04-04:
Continuous an dIncremental jogging work. Pressing i or I during any jog type stops the jog.

c-morley, 2020-04-04:
Thanks - you didn't happen to try a tandem axis machine did you?

phillc54, 2020-04-04:
> On 4 Apr 2020, at 4:26 pm, c-morley <notifications@github.com> wrote:
> 
> 
> Thanks - you didn't happen to try a tandem axis machine did you?
> 
No but I can if you need.

c-morley, 2020-04-04:
I also added fixes for the other GUIs
yes if you have time and could - I prefer someone else to test so I don't do what the functions expect.

phillc54, 2020-04-04:
OK, on a XYYZ machine with the jog slowed down so I could see what was happening better.
Pressing i or I during a jog seems to have no effect other than changing the jog rate for the next jog. The current incremental jogs completed the distance set at the start.
Doing a continuous jog with Axis buttons and pressing i or I causes a runaway jog.

c-morley, 2020-04-04:
ok I look into it more thanks.

c-morley, 2020-04-04:
I tried the sample gantry config and can't cause a runaway with the code changes. Did you recompile after the code changes? Are you using a linuxcnc gantry sim?

phillc54, 2020-04-04:
This is on a real machine. I did a fresh clone, checked out the axis_runaway_jog_fix branch and built it.
One thing I did notice, I put a print statement as the first line of the jogspeed_listbox_change function and I have yet to see it print. And I did recompile after that change.

phillc54, 2020-04-04:
I may have led you up the garden path, I was using my plasma table and I just remembered that PlasmaC destroys some widgets and rebuilds them, one being widgets.jogincr. Maybe that has some effect on the event binding. I will build a config tomorrow and try it again.

c-morley, 2020-04-04:
Ok probably sets the binding back to previous. thanks for testing i'll wait till you get a chance.

andypugh, 2020-04-04:
I am confused. Dewey posted a fix ( https://github.com/LinuxCNC/linuxcnc/commit/d7ecf9441eee55ec41e4aa59ea5c1ebb5e29548b ) and then it seems that Chris posted a different one: 
https://github.com/LinuxCNC/linuxcnc/commits/axis_runaway_jog_fix

c-morley, 2020-04-04:
Take your pick, looks like one stops jogging on increment change - one blocks increment change.

andypugh, 2020-04-04:
Well, changing incremental jog to button press is quite a big change. 
Most GUI buttons take effect on release. This lets you change your mind, drag the pointer off the button and abort the action. 
So I don't think we want to make that change in GUI behaviour.

c-morley, 2020-04-04:
well it actually put it in line with all the other jog actions. They all work on button press rather then button release. It's not that big of a change - I would say most people when incremental jogging will tap the button rather then hold it down. In the end though i guess i don;t really have strong feelings either way.

c-morley, 2020-04-04:
I forgot to mention that in that branch is fixes for the rest of the GUIs. Not sure if Dewey was working on them too

phillc54, 2020-04-05:
After removing the PlasmaC changes it appears to work as it should.
I like the fact that it jogs on key press for the GUI buttons, it now operates the same as the keyboard buttons and to me it seems more intuitive, especially when doing jogs that complete quicker than a keypress.

andypugh, 2020-04-05:
Maybe I am unusual, in that I have often pressed a GUI button (not necessarily in a LinuxCNC context) and been grateful of being able to drag the pointer off to cancel the action.
This isn't quite the same as a physical button, partly because with a mouse you can absolutely only do one thing at a time with the mouse.

c-morley, 2020-04-05:
Once you click the button, it's going to move as soon as you release. There is no way to cancel other then estop is something like that,
But also it's the only jog button like that in Axis. eg continuous jogging works on press, keyboard continuous or incremental works on press.

andypugh, 2020-04-05:
That's actually quite weird. The other momentary buttons like HOME and TOUCH-OFF support "change of mind". The jogging buttons do in incremental mode, but not in continuous mode. 

Anyway, the important thing is to have a fix.

phillc54, 2020-04-06:
I have just noticed that the spindle commands are all activated on button press as well whereas all the toolbar buttons are on button release so there is a bit of randomness there.

andypugh, 2020-04-08:
I think we can close this now.

closed 2020-04-08 by andypugh