LinuxCNC/linuxcnc #694: Run next line skip some lines after M66 command

lethang12cdt, 2020-03-02:
## Here are the steps I follow to reproduce the issue:

 1.Creating a Gcode program that has repeat loop or while loop and M66 command, see example:
; Start robot program
o101 repeat  [10]
G1 X447.5000 Y-6.0000 Z821.5000  F100001.0
G1 X70.3341 Y-441.9791 Z456.9410 F100001.0
G1 X70.3342 Y-441.9794 Z717.5789  F100001.0
M66 P1 L3 Q3
G1 X447.5400 Y-0.1134 Z365.2601  F100001.0
G1 X447.5394 Y-0.1144 Z570.4594  F100001.0
o101 endrepeat
M2

 2.Run line by line by button "run next line"
## This is what I expected to happen:
    Program run line by line
## This is what happened instead:
    Program skip some lines after M66 command

## Information about my hardware and software:

 * I am using this Linux ubuntu 16.04
 * I am using this kernel version (shown by `uname -a`): 4.9.47-rt37 PREEMPT RT
 * I am running: master version
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc): AXIS

andypugh, 2020-03-24:
I tried the code in a simulator and I don't see the problem you describe. 
I did have to change the numbers rather to make the programme fit inside the boundaries of the simulated machine. 
The feed rates are so very high that I wonder if there is even time to see the moves? 
(0.12 seconds to move from Z365 to Z 570)

lethang12cdt, 2020-04-01:
Here i make a video to show what happen:
https://drive.google.com/open?id=1wY41aVl3Y2Imq_BMP3Gtevj_YvcgK7wk
Please notice second 50th also program has repeat loop

andypugh, 2020-04-01:
Sorry, I had mis-read the description and thought this was a run-from-line problem, not a single-stepping one.

andypugh, 2020-04-01:
I tried this code with the latest master:

O100 repeat [40]
(DEBUG, 2)
G1 F100 X10
(DEBUG, 4)
G1 F100 X0
(DEBUG, 6)
M66 P1 L3 Q3
(DEBUG, 8)
G1 F100 X10
(DEBUG, 10)
G1 F100 X0
O100 endrepeat
M2

And it seemed to work as-expected. 
But I don't have any input driving the M66 input pin. Do you? 

(Something definitely looks wrong in your video, I am not arguing that point)

lethang12cdt, 2020-04-04:
> But I don't have any input driving the M66 input pin. Do you?
No, this is full simulation, no hardware connected. 

I tried your code, it only work correct from 2nd loop. See video:
https://drive.google.com/open?id=16aviPiEBHEIp_mCzawgDPVjY7RSnKcf0

And then i open another program:
https://drive.google.com/open?id=1aagJpZVG1E6OJIHrKMqjNPB2wlmsEDg4

2nd program: 

```
; Start program
o101 repeat  [10]
G64 p1
(DEBUG, 2)
G1 X47.5000 Y-6.0000 Z81.5000  F1001.0
(DEBUG, 4)
G1 X70.3341 Y-41.9791 Z46.9410 F10001.0
(DEBUG, 6)
G1 X70.3342 Y-41.9794 Z77.5789  F1001.0
(DEBUG, 8)
M66 P1 L3 Q3
(DEBUG, 10)
G1 X47.5400 Y-0.1134 Z36.2601  F1001.0
(DEBUG, 12)
G1 X47.5394 Y-0.1144 Z50.4594  F1001.0
o101 endrepeat
M2
```

This is latest master on Ubuntu 16 with realtime PREEMPT-RT

andypugh, 2020-04-04:
I have tested this sample code in 2.8 and 2.7. 
On the first run, with 2.8, the machine does not follow the programmed path. 
2.7 does not have this problem. 

#1 = 0
O100 repeat [40]
(DEBUG, 1 - #1)
G1 F50 X[2 + #1]
(DEBUG, 2 - #1)
G1 F50 X[4 + #1]
(DEBUG, 3 - #1)
M66 P1 L3 Q3
(DEBUG, 4 - #1)
G1 F50 Y[8 + #1]
(DEBUG, 5 - #1)
G1 F50 X[8 + #1]
#1 = [#1 + 8]
O100 endrepeat
M2

andypugh, 2020-04-05:
I have concluded that this issue is unrelated to M66. In fact the behaviour is far worse in my test program above with the M66 removed. 
I am closing this issue and replacing it with 
https://github.com/LinuxCNC/linuxcnc/issues/709

closed 2020-04-05 by andypugh