LinuxCNC/linuxcnc #695: Issue regarding build system of LinuxCNC

WasIstDasDingDa, 2020-03-07:
Thanks for the development of LinuxCNC. I built the software on OpenSUSE Leap 15.1 and learned about the missing dependencies in configure, make and at runtime. Most of the requirements could be met with packages of the distribution.

Only one small issue remained. The build system behaves somewhat unconventionally compared to other Linux software regarding ${PREFIX} and make install.
t's certainly not a real bug, it's obviously a feature with respect to the Debian distribution.

How can I install to e.g. /usr/local? Until the solution I have to run LinuxCNC in the place of the build tree (at §HOME), which has several disadvantages in the long term.

[linuxcnc_info.txt](https://github.com/LinuxCNC/linuxcnc/files/4301708/linuxcnc_info.txt)

andypugh, 2020-03-07:
To compile in to /usr/local you can use 
./configure -prefix=/usr/local 
See ./configure -h for more options. 

But the drawback of that is that it is hard to uninstall. You might prefer to build a debian package (.deb file) and then install that in the usual way.

WasIstDasDingDa, 2020-03-07:
Thanks for the reply. If it were that simple, I wouldn't have written.
Below are the invocations and the last lines of output, with `make install` it's all.
The statement of `make install` explicitly denies the promise of `./configure`. What is the mistake?

```
 jr@LinuxCNC:~/Entwicklung/linuxcnc/src> ./configure --enable-non-distributable=yes --with-boost-python=boost_python27 --with-realtime=/usr/realtime --prefix=/usr/local
.
.
.
######################################################################
#                LinuxCNC - Enhanced Machine Controller              #
######################################################################
#                                                                    #
#   LinuxCNC is a software system for computer control of machine    #
#   tools such as milling machines. LinuxCNC is released under the   #
#   GPL.  Check out http://www.linuxcnc.org/ for more details.       #
#                                                                    #
#                                                                    #
#   It seems that ./configure completed successfully.                #
#   This means that RT is properly installed                         #
#   If things don't work check config.log for errors & warnings      #
#                                                                    #
#   warning: If you already have an installed linuxcnc, this will    #
#         replace an existing installation.  If you have installed   #
#         a linuxcnc package, this will damage the package.          #
#   hint: To test a self-built version of linuxcnc without damaging  #
#         the package version, don't specify a --prefix              #
#                                                                    #
#   Next compile by typing                                           #
#         make                                                       #
#   then install it by typing                                        #
#         sudo make install                                          #
#                                                                    #
#   To run the software type                                         #
#         linuxcnc                                                   #
#                                                                    #
######################################################################

jr@LinuxCNC:~/Entwicklung/linuxcnc/src> make
.
.
.
make[1]: Verzeichnis „/usr/src/linux-4.14.148“ wird verlassen
cp Module.symvers *.ko ../rtlib/

jr@LinuxCNC:~/Entwicklung/linuxcnc/src> sudo make install
[sudo] Passwort für root: 
'make install' is only supported for building packages,
and it doesn't look like that is what is going on because
you have not set DESTDIR.
make: *** [Makefile:560: install] Fehler 99
```
I have compiled some software from source, but I have never encountered the term `DESTDIR`. As a test, I set `DESDIR` as an environment variable with the value of the desired installation directory (as in `--prefix=`), without success.

[config.log](https://github.com/LinuxCNC/linuxcnc/files/4302280/config.log)

P.S. Please excuse my denglish

WasIstDasDingDa, 2020-03-07:
In the meantime I have briefly dealt with staged installs ([http://buildsystem-manual.sourceforge.net/Staged-Installs.html](url)). I think it's a really good thing, but something went wrong in the build system for linuxcnc on GitHub (if not intended). I would like a DESTDIR=$HOME/inst for reference purposes. Maybe I could solve the problem myself, but I would rather mill.

andypugh, 2020-03-08:
> On 7 Mar 2020, at 18:45, WasIstDasDingDa <notifications@github.com> wrote:
> 
> Thanks for the reply. If it were that simple, I wouldn't have written.

Just a guess, but perhaps you need sudo to build in to /usr?

I am probably more confused by the build system than you are :-)

WasIstDasDingDa, 2020-03-08:
Thanks, I don't think sudo make will help, I'm gonna try it anyway. You don't walk out of your house when you're building :-) 
See you later.

WasIstDasDingDa, 2020-03-08:
I already suspected that the behaviour of the build system of LinuxCNC follows a decision of the main developer. Basically there are two options, RIP environment or regular installation via Debian package(s) in /usr. Strictly speaking there are even two partially independent build systems. I have studied the process of building the packages under Debian 10.3 in a virtual machine. For the build I followed the instructions in http://linuxcnc.org/docs/devel/html/code/building-linuxcnc.html and got 6 Debian files which could be installed with gdebi without errors. LinuxCNC runs, only the latency is bad as expected.
The only realistic way to install LinuxCNC in the /usr branch of SUSE is to involve dpkg to build the debs which then have to be converted into rpm packages. Since I do not expect the build system of LinuxCNC to be changed, the issue is closed for me at this point.
Thanks for your efforts.

Perlovka, 2020-03-11:
The problem is way more wider, than just PREFIX. It's almost impossible to build on non-debian distros, because of hardcoded paths here and there, weird install locations, and lack of documentation (dependencies etc.) I've spent 2 days just to be able to start some of example configs on gentoo.

ozzyrob, 2020-03-16:
I've installed and built Linuxcnc on Slackware and when doing a LFS & BLFS system build. I admit it was a while ago but I can't remember any serious issues during the build process. Apart from chasing and building any libs that may have been missing from either system.

WasIstDasDingDa, 2020-03-16:
@Perlovka - I think you are right
@ozzyrob  - this is of course a solution to take LinuxCNC and then build a suitable operating system around it. However, I don't think that this is what the average LinuxCNC user is looking for.

ozzyrob, 2020-03-16:
@WasIstDasDingDa WasIstDasDingDa I wouldn't call installing Slackware building a suitable operating system.
As most cnc boxes are dedicated machines why not use one of the prebuilt images ?
Some guys on the forum have reported really good results from the custom Mint images.
Or just make some space on your existing system, 32GB to 64GB is all you really need and set up a dual boot system. Saves needlessly banging your head against a wall.
If you delve into the debain directory and have a look at through the control.xxx.in files you'll get a good idea or a start of what dependencies are required, a bit of digging to suit your system will come up with the answers.
Really depends if you want to run a machine or install linuxcnc on your fave distro.

WasIstDasDingDa, 2020-03-16:
ozzyrob, please excuse and calm down. By the way, Slackware was my first Linux in 1993.  I have already installed Debian on my dedicated computer, and I know how to do it. Why do you recommend prebuilt images when you used to have LFS? Maybe someone else will come and tell me to use Windows and MACH. I know that LinuxCNC is very special, and to be able to use it on all distros would require a high effort for the developers of LinuxCNC, as well as the maintainers of the distros. And machine control is not mainstream.

saxa, 2020-03-17:
Hmm, somebody said Slackware :) I also want it on slackware. I think I build it around 2.6.x dont remember to had any big issues in doing it. Anyway I am ready to help in developing the DESTDIR option.

ozzyrob, 2020-03-17:
I was off work for 3 months with a crook back......not a lot to do, it was just a mental exercise, I think I still have all the build scripts I used.
Windows......no no no I had to use windows the other day to scan some documents........man that was a headache and that was just a 5 minute job that took 20 hahahahaha
I recommend the pre built images as it's a tried and tested method, especially if one is migrating from Windows & Mach. If you have ever browsed through the forums and seen the trouble people can get themselves into just trying to install a kernel, kernel headers and linuxcnc from packages on either a Ubuntu or Mint system, you'd see why pre built images are the way to getting up and going with no fuss.

andypugh, 2020-04-25:
I have checked and agree that --prefix does not seem to work (it used to). Even if you have --prefix and --exec-prefix too. 
Looking through the history I find: https://github.com/LinuxCNC/linuxcnc/commit/b62f3d3c7

It seems that the system was changed to default to run-in-place in 2009. 
https://github.com/LinuxCNC/linuxcnc/commit/54c202231

Do you think we can close this issue, or is a fix needed?

WasIstDasDingDa, 2020-04-26:
Thanks for walking through history. It was my guess that it must have worked at some point, as the output of the configure script and some user comments suggest.

For me as a single user, the issue is closed because I switched to Debian, the only way to do a full installation (besides on Debian derivatives).
I am not a distro zealot, and if there is certain external hardware (e.g. SDR, instrumentation, drives) with Windows-only software then I am not afraid to use Windows. If there was no LinuxCNC, I would have to use Mach for example.

However, it is a fundamental question that any free software for Linux should be able to be built on any Linux platform, regardless of the package system and other details. And this requirement is not achieved with the current build system of LinuxCNC. The fulfillment of the dependencies is completely in the responsibility of the user, but the author/developer/maintainer should document the dependencies in detail.

So do what you think is right.

ozzyrob, 2020-04-26:
Actually, going back to Slackware, the user does have to look for required libs, sometimes the devs provide the requirements, other times you have to go through the configure process a few times.
Thanks to SlackBuilds.org building some packages have been made easier, but mostly there is still a lot of hunting to do. Building gitlab on my Slackware server was fun hahahaha.
The Debian build infrastructure does make package building somewhat easier......if the developer has put together the correct debian dir structure. Looking through the debian/control files can help in finding out which other packages are required.
Maybe the whole Linux world needs a bit of a shake up........maybe a common package format and method of building packages that is distro agnostic.
But in all honesty the project would require more users form other distros that have a high level of ability and knowledge for their particular distro. Which would be great.

WasIstDasDingDa, 2020-04-26:
@ozzyrob , I agree with you. If I had more time, knowledge (and desire), I would make a fork of LinuxCNC. But only to simplify the "porting" to other Linux distributions. I would start with OpenSUSE and I would invite you for the part about Slackware. For Gentoo there was already a knowledgeable user (Perlovka) in the thread. The work would essentially be reverse engineering and puzzling. You don't have to be a great C programmer but you should be familiar with Shell programming, the GNU Build System and preferably different package systems.

andypugh, 2020-04-27:
From the mailing list: 
https://sourceforge.net/p/emc/mailman/message/36993723/

WasIstDasDingDa, 2020-04-27:
@andypugh Thanks, at last you have done what could be done, also in the thread on NTLINUX/RTAI. I am completely satisfied with the result. I am running a recent LinuxCNC on a recent operating system with a recent RTAI kernel. Everything else falls into the category "nice to have". You can close the issue.

closed 2020-05-11 by andypugh