LinuxCNC/linuxcnc #699: Segmentation fault with G72 command

papagno-source, 2020-03-18:
## Here are the steps I follow to reproduce the issue:

 1.start config axis/lathe.ini
 2.load 
 3.load program :
G54 G18 G21 F400
G40 T2M6G43
m3 s100
o100 sub
g1 x5 f1000
o100 end sub
G72 Q100 X-5 Z10 I1 R0.1
m30

## This is what I expected to happen:
[ 9050.939469] axis[6851]: segfault at 0 ip b6023afa sp bfd45b1c error 4 in librs274.so.0[b5f1e000+194000]

## Information about my hardware and software:

* [ ] A binary I built myself
Actual master 2.9.pre0
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):
AXIS, only simulation

mark-v-d, 2020-03-18:
See pull request #700

papagno-source, 2020-03-18:
another the same segmentation fault :

G54 G18 G21 F400
G40 T2M6G43
m3 s100
o100 sub
g1 z0f1000
x60
o100 end sub
G72 Q100 X60 Z0  I1 R0.1
m30

andypugh, 2020-03-18:
Did you pull and rebuild?

mark-v-d, 2020-03-18:
I added pull request #701 to fix this, thanks for testing this stuff :-)

closed 2020-04-04 by andypugh