LinuxCNC/linuxcnc #702: Contour ShuttlePro Not Recognized

WhyGitHub1, 2020-03-21:
## Here are the steps I follow to reproduce the issue, and my solution that now works:

 1. I was using a ShuttleExpress and it was working fine in V2.9.0 pre0 but I wanted more buttons for 5 axis machine I was building up from an old 4 axis.
 2. I modified the Hal file per linuxcnc documents, and hooked up a Contour Design ShuttlePro . It was not recognized by LinuxCNC
 3. I checked the src file shuttle.c noted that the vendor_id and product_id do not match when I run evtest on the shuttlepro.  I get 0xb33 and 0x30 respectively.
4. I today pulled src from git, set up a build environment, modified the shuttle.c file "shuttlepro" vendor_id and product_id to 0x0b33 and 0x0030 respectively.
5. I then built and tested this on my home built 5-axis machine, and with one more tweak, "see item 6 below", all 5 axis worked.
6. Note that I also found that the documentation which says use shuttle.pro.2.button-0 did not work.  I changed this to shuttle.pro.0.button.0 etc for each button in my .hal file for the shuttle, and this worked.
7. Note: For 5 axis I used the second row of buttons 4-8 to (5 buttons) to control 5 axis.   All 5 axis jog on actual hardware using shuttlepro with this change.

Perhaps there is another version of the shuttle that has these ids.  Mine is from Contour Design and is ShuttlePro V.2.  snippet of my code change below.

contour_dev_t contour_dev[] = {
    {
        .name = "shuttlexpress",
        .vendor_id = 0x0b33,
        .product_id = 0x0020,
        .num_buttons = 5,
        .button_mask = { 0x0010, 0x0020, 0x0040, 0x0080, 0x0100 }
    },

    {
        .name = "shuttlepro",
        .vendor_id = 0x0b33,  //my change
        .product_id = 0x0030, //my change
        .num_buttons = 13,
        .button_mask = { 0x0001, 0x0002, 0x0004, 0x0008, 0x0010, 0x0020, 0x0040, 0x0080, 0x0100, 0x0200, 0x0400, 0x0800, 0x1000 }
    }

andypugh, 2020-04-07:
The vendor ID currently in the driver is actually "PI Technology" whereas the one you have (0x0b33) is Contour Design, so your change looks good. But the other one must have worked at some point...
I have added (rather than replaced) your data to the driver in 2.8 and 2.9. 
Could you please test that the driver now works for you? I have no hardware to test with.

WhyGitHub1, 2020-04-07:
Hi,
I’ll give it a try and let you know.

WhyGitHub1, 2020-04-07:
I downloaded and built from github this afternoon.  Initially did not work, but looked at the shuttle.c source and noted that you called the Contour Design  as shuttleproV2.  I updated my .hal file to shuttleproV2.0.button# etc, and yes this change is working for my 5-axis.  xyzbc.

Thanks

andypugh, 2020-04-07:
shuttleproV2.0...... is an unwieldy hal pin name. I feel that it would make a lot more sense for all shuttleexpress and shuttlepro pins to just be called "shuttle".
The config conversion script can handle this. 

Would it be possible to test the new amdypugh/shuttle branch of LinuxCNC where this change has been made to check I haven't broken anything?

WhyGitHub1, 2020-04-07:
I’ll see if I can remember/figure out how to switch the branch and give it a go.

WhyGitHub1, 2020-04-08:
Hi,
I cloned -b andypugh/shuttle to a different dev directory and built it.  Changed Hal file to shuttle.0.button.  Build is a 2.8pre. Vs 2.9.  Doesn’t work, shuttle.c still has proV2.
Let me know if I grabbed the right files. Built fine just doesn’t like the shuttle.

On Apr 7, 2020, at 6:52 PM, John Hodapp <bigshot@msn.com> wrote:

I’ll see if I can remember/figure out how to switch the branch and give it a go.

WhyGitHub1, 2020-04-08:
Believe it failed on check for the 1st button in my hal list

On Apr 8, 2020, at 1:25 PM, John Hodapp <bigshot@msn.com> wrote:

Hi,
I cloned -b andypugh/shuttle to a different dev directory and built it.  Changed Hal file to shuttle.0.button.  Build is a 2.8pre. Vs 2.9.  Doesn’t work, shuttle.c still has proV2.
Let me know if I grabbed the right files. Built fine just doesn’t like the shuttle.

On Apr 7, 2020, at 6:52 PM, John Hodapp <bigshot@msn.com> wrote:

I’ll see if I can remember/figure out how to switch the branch and give it a go.

andypugh, 2020-04-08:
I missed a change (I would think it probably failed with the "counts" line? Which is why we need to check) 
I updated the branch. 
(git pull then make should build the new version) 

A quick and easy check is:

```
halrun
loadusr shuttle
show pin
exit
```

WhyGitHub1, 2020-04-08:
Yes that works.  Shows pins.  Also ran LinuxCNC with shuttle.0.button-# etc. All axis buttons I have programmed work along with the inner and outer wheels. V2.8.0~pre1

WhyGitHub1, 2020-04-08:
Oh, and thanks !

On Apr 8, 2020, at 4:59 PM, John Hodapp <bigshot@msn.com> wrote:

Yes that works.  Shows pins.  Also ran LinuxCNC with shuttle.0.button-# etc. All axis buttons I have programmed work along with the inner and outer wheels. V2.8.0~pre1